md-raid-utilities/ledmon #9: Segmentation fault in ledmon  --list-controller

reflectored, 2018-01-12:
We have renamed the filename, did not touch the ledmon implementation.
The backtrace is below. I can send the dump if necessary.

(gdb) r --list-controller
Starting program: /root/./quobyte_ledmon --list-controller
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
/sys/devices/pci0000:00/0000:00:11.4 (AHCI)
/sys/devices/pci0000:00/0000:00:1f.2 (AHCI)

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7837694 in vfprintf () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff7837694 in vfprintf () from /lib64/libc.so.6
#1  0x00007ffff7840879 in printf () from /lib64/libc.so.6
#2  0x0000000000404fed in cntrl_print_all (ctrl_list=0x6133dc) at cntrl.c:443
#3  0x000000000040266b in _cmdline_parse (argc=2, argv=0x7fffffffe4b8) at ledmon.c:446
#4  0x00000000004031b1 in main (argc=2, argv=0x7fffffffe4b8) at ledmon.c:810

mzylowski, 2018-01-12:
Can you give some more details about hardware connected to your machine? Is there any hardware card etc? Is there any controller missing in given output?

reflectored, 2018-01-12:
There are 2 SAS3008 controllers in this enclosure, with 4 links (4 enclosure devices are present).

The controllers that are missing:
H/W path           Device     Class          Description
========================================================
/0/100/1/0         scsi10     storage        SAS3008 PCI-Express Fusion-MPT SAS-3
/0/1/0                scsi11     storage        SAS3008 PCI-Express Fusion-MPT SAS-3

mzylowski, 2018-01-16:
@reflectored 
In #10 you can check some improvements related with --list-controllers option. Now it is a part of ledctl app. Can you check this patch?
If you're still reproducing this issue, could you attach for us core dump file related with problem?

reflectored, 2018-01-17:
We have checked it, it works! Thank you!

closed 2018-01-24 by reflectored