md-raid-utilities/ledmon #12: SES is not used for enclosure device that are directly connected.

reflectored, 2018-02-23:
We have a controller that has enclosure devices connected without expanders,  they have to be controlled with SES and they cannot be controlled with smp.

The issue is in this line:
https://github.com/intel/ledmon/blob/8cb8c5fe04afbb328a61115c7d5bcc639419110d/src/block.c#L89
and this function:
https://github.com/intel/ledmon/blob/8cb8c5fe04afbb328a61115c7d5bcc639419110d/src/block.c#L55

If it finds no **"/expander"** in the path, it decides to use the function: **scsi_smp_fill_buffer** for this device. I think you should check if its an enclosure device differently not based solely on that string.

mzylowski, 2018-03-29:
@reflectored 
Can you tell us what exactly hardware do you use? I will try to find something similar to perform testing.

reflectored, 2018-03-29:
@mzylowski 
The hardware was a HPE Apollo 4200 server, we made our own fork to support it. I have previously suggested our workaround as a pull request: https://github.com/reflectored/ledmon/blob/ses_support_fix/src/cntrl.c#L219-L220

reflectored, 2018-07-12:
@mzylowski hello there. Is there any new information about this?

mzylowski, 2018-07-12:
Hello!
We cannot find any hardware in our LAB for testing our changes around this. That is main problem.
Currently we looking in some other teams.
As I understand something like this will be fine: https://www.hpe.com/us/en/product-catalog/servers/apollo-systems/pip.hpe-apollo-4200-gen9-server.8261831.html ?

Is there any possibility to receive dump of your sysfs from "HPE Apollo 4200 server"? It would be really helpful. 

Also maybe you want rebase your proposal against current master (after our SES refactoring from v.090)? 

We can try review it again and trigger our test suite to be sure it doesn't break anything.

Sorry for long delay.

reflectored, 2018-08-15:
Hello Michał,

We have moved to use SES directly, which is also used by ledctl, therefore this issue is rendered obsolete.

Thank you for your help!

closed 2018-08-15 by reflectored