md-raid-utilities/ledmon #19: ledctl: /dev/nvme0n1: device not supported

vermorth, 2018-09-17:
I want to use ledctl to locate my nvme whose model is `INTEL SSDPE2KE020T7`. But ledctl reported device not supported. Why?

```
[root@100 ~/rpms]# ledctl locate=/dev/nvme0n1
ledctl: /dev/nvme0n1: device not supported
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.
```

```
[root@100 ~/rpms]# nvme list   
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     PHLE8066001Y2P0DGN   INTEL SSDPE2KE020T7                      1           2.00  TB /   2.00  TB    512   B +  0 B   QDV10170
/dev/nvme1n1     PHLE806600442P0DGN   INTEL SSDPE2KE020T7                      1           2.00  TB /   2.00  TB    512   B +  0 B   QDV10170
```

```
[root@100 ~/rpms]# ledmon -v
Intel(R) Enclosure LED Monitor Service 0.90
Copyright (C) 2009-2018 Intel Corporation.

This is free software; see the source for copying conditions. There is NO warranty;
not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

pbaldysi, 2018-09-27:
Hi,
For LED management for NVMe device there is a VMD hardware in CPU needed.
https://www.intel.com/content/www/us/en/architecture-and-technology/intel-volume-management-device-overview.html
It looks like your NVMe devices are directly connected to CPU (without VMD). 
To confirm that - could you paste sysfs paths for those NVMes (ll /sys/block)?

Thanks
Pawel

vermorth, 2018-09-28:
Thank you for your reply.
ll /sys/block: 
```
[root@localhost~]# ll /sys/block | grep nvme
lrwxrwxrwx 1 root root 0 Sep 11 21:03 nvme0n1 -> ../devices/pci0000:80/0000:80:02.2/0000:85:00.0/nvme/nvme0/nvme0n1
lrwxrwxrwx 1 root root 0 Sep 11 21:03 nvme1n1 -> ../devices/pci0000:80/0000:80:02.3/0000:86:00.0/nvme/nvme1/nvme1n1
```

pbaldysi, 2018-10-01:
Hi,
Your log confirms that you have setup without VMD.
Ledmon supports LED blinking for NVMe - only with the drives connected through VMD.

Example output for VMD attached drives - drives are exposed as connected through additional PCI domain that is larger then 0xFFFF (10000 with my example):
```
lrwxrwxrwx 1 root root 0 Oct  1 04:49 nvme0n1 -> ../devices/pci0000:17/0000:17:05.5/pci10000:00/10000:00:00.0/10000:01:00.0/10000:02:00.0/10000:03:00.0/nvme/nvme0/nvme0n1
lrwxrwxrwx 1 root root 0 Oct  1 04:49 nvme1n1 -> ../devices/pci0000:17/0000:17:05.5/pci10000:00/10000:00:00.0/10000:01:00.0/10000:02:01.0/10000:04:00.0/nvme/nvme1/nvme1n1
```

Thanks
Pawel

closed 2018-10-01 by pbaldysi