md-raid-utilities/ledmon #29: compile error of format not a string literal and no format arguments

2005songliwei, 2019-04-02:
Hi,
I came across an compile error in util.c since commit 
dfa5e6620d9c ("[ledmon] Unify logging methods.")

error is:

utils.c: In function '_log':
utils.c:314:21: error: format not a string literal and no format arguments [-Werror=format-security]
   fprintf(s_log, lli->prefix);

I try to add an "%s" to it, then the error is not produced:
   fprintf(s_log, "%s", lli->prefix);

Is it an issue? Could you help check?

Thanks,
Liwei.

mdabrows, 2019-04-02:
Hello,

this issue has been fixed by bee7dee so it looks that your working branch is outdated. Could you rebase your branch to current master and test if this issue can be reproduced?

Regards,
Mariusz

2005songliwei, 2019-04-03:
HI Mariusz,

Thanks for your input, after update to the newest version, I got another two same compile error:

| utils.c: In function '_log':
| utils.c:302:19: error: format not a string literal and no format arguments [-Werror=format-security]
|     fprintf(s_log, msg);
|                    ^~~
| utils.c:306:25: error: format not a string literal and no format arguments [-Werror=format-security]
|    syslog(lli->priority, msg);
|                          ^~~
| cc1: some warnings being treated as errors

after add "%s" to all fprintf and syslog this error disappeared.

                        _log_timestamp();
                        fprintf(s_log, "%s", lli->prefix);
                        fprintf(s_log, "%s", msg);
                        fprintf(s_log, "%s", END_LINE_STR);
                        fflush(s_log);
                }
                syslog(lli->priority, "%s", msg);

Thanks,
Liwei.

mtkaczyk, 2019-04-03:
Hello,
Can you retest it with #30 ?

Thanks,
Mariusz

2005songliwei, 2019-04-04:
Yes, it can fix the issue now.

Thanks,
Liwei.

PR #30 (2019-04-04): Add format argument
Add missing format argument to fix format-security issues.

Fixes #29
closed 2019-04-04 by 2005songliwei