md-raid-utilities/ledmon #35: ledctl dont change LED status

crusaderflex, 2019-05-16:
Hi, 

i have a Supermicro Mainboard H11DSi-NT with Supermicro Backplane BPN-SAS3-743A-N4
I hoped the new amd-sgpio patch would work an I could change die LEDs of the
Backplane, but nothing changed. Not sure if my SATA-Controller of the MB is supported.
The activity LED is blinking blue, if the softraid writes to disk and at  PC start i can see
a red LED flashing very short. So I think the cable ist right. 

# **ledctl -L**
/sys/devices/pci0000:40/0000:40:08.1/0000:42:00.2 (AMD SGPIO)
/sys/devices/pci0000:50/0000:50:08.1/0000:52:00.2 (AMD SGPIO)
/sys/devices/pci0000:10/0000:10:08.1/0000:14:00.2 (AMD SGPIO)

# **ledctl locate=/dev/sda**
/dev/shm/ledmon.conf: does not exist, using global config file
ledctl: AMD Drive: port 2, ata port 10, drive bay 2, initiator 0
ledctl: AMD Drive: port 7, ata port 17, drive bay 1, initiator 1
ledctl: AMD Drive: port 1, ata port 1, drive bay 3, initiator 0
ledctl: (raid_device_init) path: md2, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (raid_device_init) path: md0, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (raid_device_init) path: md1, level=2, state=5, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl:
ledctl: Setting LOCATE_OFF...
ledctl:         device: .../ata12/host11/target11:0:0/11:0:0:0/block/sdb
ledctl:         buffer: .../ata17/host16/scsi_host/host16/em_buffer
ledctl: AMD Drive: port 2, ata port 12, drive bay 2, initiator 0
ledctl: AMD Register: 00100030 00c08240 00000001 00000060
ledctl: SGPIO Header
ledctl:            message type: 3                 data size: 0
ledctl:            message size: 10
ledctl: SGPIO Request Register:
ledctl:              frame type: 40                 function: 82
ledctl:           register type: c0           register index: 0
ledctl:          register count: 1
ledctl: SGPIO AMD Register
ledctl:               initiator: 0                  polarity: 0
ledctl:           bypass enable: 1          return to normal: 1
ledctl: CFG Register: 00140030 00008240 00000002 00800000 00210000
ledctl: SGPIO Header
ledctl:            message type: 3                 data size: 0
ledctl:            message size: 14
ledctl: SGPIO Request Register:
ledctl:              frame type: 40                 function: 82
ledctl:           register type: 0            register index: 0
ledctl:          register count: 2
ledctl: SGPIO Configuration Register:
ledctl:                 version: 0         gp register count: 0
ledctl:      cfg register count: 0              gpio enabled: 1
ledctl:             drive count: 0          blink gen rate A: 0
ledctl:        blink gen rate B: 0        force activity off: 2
ledctl:         max activity on: 1      stretch activity off: 0
ledctl:     stretch activity on: 0
ledctl: TX Register:  00100030 00038240 00000001 a0a0a0a0
ledctl: SGPIO Header
ledctl:            message type: 3                 data size: 0
ledctl:            message size: 10
ledctl: SGPIO Request Register:
ledctl:              frame type: 40                 function: 82
ledctl:           register type: 3            register index: 0
ledctl:          register count: 1
ledctl: SGPIO TX Register:
ledctl:         drive 0: error 0, locate 0, activity 5
ledctl:         drive 1: error 0, locate 0, activity 5
ledctl:         drive 2: error 0, locate 0, activity 5
[... 125 lines cut ...]
ledctl:              frame type: 40                 function: 82
ledctl:           register type: 3            register index: 0
ledctl:          register count: 1
ledctl: SGPIO TX Register:
ledctl:         drive 0: error 0, locate 0, activity 5
ledctl:         drive 1: error 0, locate 0, activity 5
ledctl:         drive 2: error 0, locate 0, activity 5
ledctl:         drive 3: error 6, locate 0, activity 6

# **ledctl -x**
/dev/shm/ledmon.conf: does not exist, using global config file
ledctl: AMD Drive: port 2, ata port 10, drive bay 2, initiator 0
ledctl: AMD Drive: port 7, ata port 17, drive bay 1, initiator 1
ledctl: AMD Drive: port 1, ata port 1, drive bay 3, initiator 0
ledctl: (raid_device_init) path: md2, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (raid_device_init) path: md0, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (raid_device_init) path: md1, level=2, state=5, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: missing operand(s)... run ledctl --help for details.
ledctl: main(): _ibpi_parse() failed (status=STATUS_IBPI_DETERMINE_ERROR).

mdabrows, 2019-05-16:
Hello,
@nfont could you take a look at this issue?

Regards,
Mariusz

nfont, 2019-05-20:
@crusaderflex I am not seeing anything wrong with the values that ledmon is writing to the SGPIO registers. One thought would be that the BIOS on your SuperMicro system doesn't support SGPIO LED control.

crusaderflex, 2019-05-22:
@nfont Thank you for your answer. Good to known, that the registers are right.
I ask my supplier to request at Supermicro if Mainboard, Backplane an Cable supports SGPIO LED control.
If i have an answer i will give feedback here.

pbaldysi, 2019-05-29:
Since this is not a defect in ledmon - closing this one.

closed 2019-05-29 by pbaldysi