md-raid-utilities/ledmon #47: compilation errors because of references to packed struct members

jsynacek, 2019-10-21:
With gcc 9.2.1, I'm getting a lot of these during the compilation of ledmon-0.93:
```
...
amd_sgpio.c:482:18: error: taking address of packed member of 'struct transmit_register' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  482 |  _dump_sgpio_req(&tx_reg->req);
      |                  ^~~~~~~~~~~~
amd_sgpio.c: In function '_write_tx_register':
amd_sgpio.c:489:18: error: taking address of packed member of 'struct transmit_register' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  489 |  _init_sgpio_hdr(&tx_reg->hdr, 0, sizeof(*tx_reg));
      |                  ^~~~~~~~~~~~
amd_sgpio.c:490:18: error: taking address of packed member of 'struct transmit_register' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  490 |  _init_sgpio_req(&tx_reg->req, 0x40, 0x82, SGPIO_REQ_REG_TYPE_TX, 0, 1);
      |                  ^~~~~~~~~~~~
amd_sgpio.c: In function '_dump_amd_register':
amd_sgpio.c:543:2: error: converting a packed 'struct amd_register' pointer (alignment 1) to a 'uint32_t' {aka 'unsigned int'} pointer (alignment 4) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  543 |  uint32_t *reg = (uint32_t *)amd_reg;
      |  ^~~~~~~~
amd_sgpio.c:144:8: note: defined here
  144 | struct amd_register {
      |        ^~~~~~~~~~~~
amd_sgpio.c:548:18: error: taking address of packed member of 'struct amd_register' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  548 |  _dump_sgpio_hdr(&amd_reg->hdr);
      |                  ^~~~~~~~~~~~~
...
```
To be honest, I don't know if such references are even supposed to work, but I can imagine that the generated code might act funny on architectures like ppc64.

mdabrows, 2019-10-22:
@nfont could you take a look on this issue? -Waddress-of-packed-member is enabled by default with gcc-9, so it seems that amd_sgpio code has to be refactored to not use pointers to members of packed structures.

jsynacek, 2019-10-23:
I've consulted this with people who know more than I do about this topic and I was told that the code itself should be safe, as architectures that do not support unaligned memory accesses are rare. However, I still think it's better to not use them, unless there is a really strong reason to do so, which I can't even think of and, in my opinion, is not the case here.

nfont, 2019-10-25:
I can taker a look at updating the amd code. The use of the packed structures were to keep the compiler from padding any of the structures so they can be written directly to the sysfs file.

jsynacek, 2019-10-30:
The use of packed structures is not a problem. The problem is the use of references to the members of those structures in function calls. Having a function(s) that take(s) a reference to the entire structure and then deal(s) with the members inside should be fine.

mdabrows, 2019-11-29:
Hello @nfont do you have any progress on this issue?

Regards,
Mariusz

nfont, 2019-12-02:
I have code that I have been testing and will get a pull request put together this afternoon.

nfont, 2019-12-02:
I opened pull request #52 to resolve this issue.

closed 2019-12-06 by rsobans