md-raid-utilities/ledmon #49: No Enclosure Script

DanielLester83, 2019-10-31:
Here's a simple script I made to create activity to identify a failed HDD in a zfs pool where the disks are directly connected to the motherboard thus no enclosure and ledctl won't work. 
Requires zpool, awk, & ioping

sudo zpool status TANK | awk '$2 ~ /^FAULTED$/ {system("sudo ioping /dev/disk/by-id/" $1)}'

where TANK is the zpool tank name

mtkaczyk, 2019-12-03:
Hello,
I don't see any issue in your description. Can you explain it more?

Thanks, 
Mariusz

DanielLester83, 2019-12-03:
There's no real issue, I was just posting the code for posterity and in case you wanted to integrate it and expand the usability of your software.

mtkaczyk, 2019-12-04:
Hi,
We don't plan to support directly attached HDDs or zfs.
You can implement it and open pull request if you want.
For now I'm closing the issue.

Thanks,
Mariusz

closed 2019-12-04 by mtkaczyk