md-raid-utilities/ledmon #58: ledmon with zfs

syswombat, 2020-01-27:
hello
i have proxmox installation 4 disk connected on the Motherboard with sata.
i know the sata port but should find out the sda port

in my case i see the sda, sdb sdc sdd but the led will not flash maybe because the Disk are in a IcyDock - FatCage MB153SP-B?

# ledctl -L
/sys/devices/pci0000:00/0000:00:17.0 (AHCI)

i only have one maybe should be 4 entry for each disk, 

have a nice day
vinc

mdabrows, 2020-01-29:
Hello,

"ledctl -L" output is fine - it lists controllers, not disks. Your motherboard probably has 1 SATA controller which is shown by ledctl.
I've made a quick research about your HDD cage and it looks that it doesn't support status LED but only activity LED.

Regards,
Mariusz

mtkaczyk, 2020-09-02:
No input from January.
Closing.

closed 2020-09-02 by mtkaczyk