md-raid-utilities/ledmon #63: ledctl overrides ibpi request under certain conditions

ntnx-alay, 2020-04-18:
Not using ledmon, just using ledctl to manage LED status of one of the soft raid devices.

ledctl is expected to update status of LED with user requested irrespective of current RAID conditions.

When raid constructions is "In progress" and requested to set to Normal, ibpi requested status gets overridden by value regarding to "DEGRADED".

ntnx-alay, 2020-04-22:
Potential non-intrusive fix that prevents override.
#64

mtkaczyk, 2020-04-28:
Hello,
Please try set "locate_off" instead "normal". It shouldn't be overwritten and later is translated into normal.
As I describe in  #64 there is a plan for adding communication between ledmon and ledctl.  It will fix  the issue because ledctl will becomes simple updater without additional determining logic.
For now it is a bug and should be still open.

Thanks,
Mariusz

ntnx-alay, 2020-04-29:
Hello Mariusz,

You are right, I used locate_off and it does not overwrite requested - I think due to enum value of locate_off is higher than degraded.

Thank you for suggestion and workaround.

mtkaczyk, 2020-05-19:
Hi @ntnx-alay,
Could you verify #67?
This pull forces user request to be set instead of state with the highest priority.

ntnx-alay, 2020-05-20:
Confirmed #67 addresses use model.

mtkaczyk, 2020-05-20:
Thanks for your support.

Mariusz

closed 2020-05-20 by mtkaczyk