md-raid-utilities/ledmon #74: ledctl: Unable to locate all disks in md devices

NHellFire, 2020-09-22:
The docs contain an [example](https://github.com/intel/ledmon/blob/1d4a19/doc/ledctl.pod#examples) for locating all disks in an md device, however this does not work:

```
root:~# ledctl --debug locate=/dev/md0
ledctl: /dev/md0: device not supported
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.
```

Other info:
```
root:~# ledctl --version
Intel(R) Enclosure LED Control Application 0.94 
Copyright (C) 2009-2019 Intel Corporation.

This is free software; see the source for copying conditions. There is NO warranty;
not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
root:~# ledctl --debug -L
ledctl: (raid_device_init) path: md0, level=6, state=6, degraded=0, disks=9, type=1
ledctl: (_set_block_state): device: sdm, state: NORMAL
ledctl: (_set_block_state): device: sdl, state: NORMAL
ledctl: (_set_block_state): device: sdk, state: NORMAL
ledctl: (_set_block_state): device: sdj, state: NORMAL
ledctl: (_set_block_state): device: sdi, state: NORMAL
ledctl: (_set_block_state): device: sdh, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
ledctl: (_set_block_state): device: sdf, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
/sys/devices/pci0000:00/0000:00:02.0/0000:01:00.0 (SCSI)
/sys/devices/pci0000:00/0000:00:15.0/0000:06:00.0 (SCSI)
```

OS is Ubuntu 20.04, kernel 5.4

mtkaczyk, 2020-11-17:
Hi,
Looks like a bug:

> A device is a path to file in /dev directory or in /sys/block directory.  It may identify a
> block device, a RAID device or a container device.  In case of a RAID device or a container
> device a state will be set for all block devices associated, respectively.

Thanks for your report.
Someone will look into it.
You can propose fix if you want to contribute.

Thanks,
Mariusz

mtkaczyk, 2021-04-15:
Hello,
We agreed to drop this feature and remove it from documentation.
Do you have any objections?

Mariusz

mtkaczyk, 2021-04-23:
@NHellFire ping?
If you don't give a feedback, I will merge the changes next week.

Thanks!

closed 2021-04-29 by mtkaczyk

NHellFire, 2021-04-30:
Fine for me, I'm no longer using mdadm and my stated workaround should help
others.

PR #80 (2021-04-29): Update manual
Remove legacy support for bliking via md raids from manual.
Update examples section to make it more complex.

Fixes #74