md-raid-utilities/ledmon #75: Define Ledmon behavior for corner cases

mtkaczyk, 2020-10-28:
# Problem description
For some corner case scenarios Ledmon behavior is not obvious and cannot be easily
determined. The goal is to decide how ledmon should behave in those scenarios.  
This list contains known undefined behaviors for the moment of submitting. I strongly
believe that it is a good point to start. Our expectation is to define rules
which ledmon should follow. Those rules will be noted and saved in the
documentation. This new file will be added to repository. Required changes will
be applied to product later.

## Scenario 1: Manual stop / RAID deactivation

### Problem:
 When RAID is stopped, Ledmon marks all disks as normal. It does not check the reason.
 It may be caused by:
 - user, if raid stop is requested
 - system, if it is caused by drive failure or IO errors.

From Ledmon perspective there - there is no good way to tell the difference. 

### Question:
 If RAID array disappears, should member LEDs stay normal or blink failure?

### Steps to reproduce:
> **Manual stop:**  
> Create container with 2 disks:  
> #mdadm -CR $CONTAINER_NAME -e imsm -n2 $DISK1 $DISK2  
> Create RAID:  
> #mdadm -CR $RAID_NAME -l1 -n2 /dev/$DISK1 /dev/$DISK2  
> Stop created RAID:  
> #mdadm -S /dev/md/$RAID_NAME  
>
> **System stop:**  
> Create container with 2 disks:  
> #mdadm -CR $CONTAINER_NAME -e imsm -n2 $DISK1 $DISK2  
> Create RAID:  
> #mdadm -CR $RAID_NAME -l0 -n2 /dev/$DISK1 /dev/$DISK2  
> Stop RAID by system – remove one drive:  
> #echo 1 > /sys/block/$DISK1/device/delete

### Actual result
RAID is removed from system (see /proc/mdstat). If one drive is removed then it is
marked as failed by Ledmon. Other drives are marked as normal.

##  Scenario 2: Matrix RAID – remove one volume.

### Problem
Up to two volumes can be created upon one IMSM container. Each array can have different
failure tolerance. As a result one volume can be gone, but second one is still
active. It is a more complicated case for **Scenario 1**.

### Question
Should Ledmon mark failure on each array member or only on failed members?

Steps to reproduce:

> Create container with 2 disks:  
> #mdadm -CR $CONTAINER_NAME -e imsm -n2 $DISK1 $DISK2  
> Create partial capacity RAID1:  
> #mdadm -CR $RAID_NAME_1 -l1 -n2 /dev/$DISK1 /dev/$DISK2 -z 10G  
> Create RAID0 from rest of free space :  
> #mdadm -CR $RAID_NAME_2 -l0 -n2 /dev/$DISK1 /dev/$DISK2  
> Remove one drive or stop one volume:
> - #echo 1 > /sys/block/$DISK1/device/delete
> - #mdadm -S /dev/md/$RAID_NAME_2

### Actual result
If DISK_1 is removed then it is marked as failed. RAID0 disappears, RAID1 is
degraded. DISK_2 is marked as normal.

## Scenario 3: Remove mounted volume

### Problem
If volume is in use then mdadm is not able to unmount it before stop. If
array fails, then - as a last resort - udisk is triggered. The root cause is the same as for **Scenario 1** and **Scenario 2** - array becomes failed but in this case
RAID won't be stopped. Ledmon can't detect that array is failed, it relies on
md device existing.

### Question
How Ledmon should behave when it is not possible to stop RAID when it becomes failed?

### Steps to reproduce:
> Create container with 2 disks:  
> #mdadm -CR $CONTAINER_NAME -e imsm -n2 $DISK1 $DISK2  
> Create RAID:  
> #mdadm -CR $RAID_NAME -l0 -n2 /dev/$DISK1 /dev/$DISK2  
> Mount RAID 
> #mkfs /dev/md/$RAID_NAME  
> #mount /dev/md/$RAID_NAME /mnt  
> Generate IO:  
> #dd if=/dev/urandom of=/dev/md/$RAID_NAME &  
> During IO remove one drive:  
> #echo 1 > /sys/block/<disk1>/device/delete

### Actual result
Ledmon marks removed drive as failure. Second drive from this RAID is marked as normal.

# Summary
Those three scenarios have one common result, in each case RAID
becomes unusable because it is failed or disappears from system. We should define
and document the behavior which should be expected in this case.  
Marking as failure looks reasonable, but it can be misleading in some cases. From
other hand marking as normal won't inform about any raid failure. We want to know
user expectation and provide best solution.

therealjuanmartinez, 2021-05-13:
What is the behavior when there is a hot unplug for a drive that is NOT in a RAID array?  Is it still marked as failure?

mtkaczyk, 2021-05-14:
Each drive removal results with failure led unless drive is excluded in ledmon config.

alpinus4, 2021-11-23:
We decided to avoid the problem by limiting the scope of ledmon behaviour.
RAID states (such as Normal, Degraded, Failed) shall not have an effect on LED pattern,
with an exception of rebuild, initialization or migration.

Decided behaviour:

| LED pattern | When |
| --- | --- |
| None | Disk in normal state, RAID without any operations in the background such as rebuild, initialization or migration |
| Failed | Disk has a failed state |
| Rebuild | Rebuild, initialization or migration is being done in the background on RAID |
| Locate | Manually activated by the user |

closed 2021-11-23 by mtkaczyk