md-raid-utilities/ledmon #76: Clarify supported hardware configurations of LEDs

Ashark, 2020-11-20:
Manual is not very clear about which cases are supported.

I have recently bought OImaster MR-6601 (Thermaltake MAX 2506) sata bay, and I was exploring if it is possible to activate locate blinking on it. Activating locate pattern with `ledctl locate=/dev/sda` did nothing.

The sata bay looks looks like this:
[image]

There are two LEDs for each bay: power (blue) and access (red). I thought that Access led and Status led are synonyms. But looks like it is not, according to what I have read in #58.

The man page for ledctl says:
`There are two types of systems: 2-LEDs systems (Activity LED , Status LED ) and 3-LEDs systems (Activity LED , Locate LED , Fail LED ).`
So can you clarify about led namings on bays like mine?
Activity led - am I right that it is led which blinks when you write/read data from the drive?
Status led - am I right that it is Power led? Or what is status?
Access led - am I right it is the same as Activity led?

So my bay is 2-led? If not, then why there is no mentioning in man about such bays with one led (Access Led, power LED is not counted it seems). May it be added to man page?

And another statement which is not very clear to understand to me:
`LED management ( AHCI ) and SAF-TE protocols are not supported.`
Why it says led management not supported? I have In `ledctl -L` this output `/sys/devices/pci0000:00/0000:00:17.0 (AHCI)`. So is my controller supported?

mtkaczyk, 2020-11-23:
Hi,
> There are two LEDs for each bay: power (blue) and access (red). I thought that Access led and Status led are synonyms. But > looks like it is not, according to what I have read in #58.
The man page for ledctl says:
There are two types of systems: 2-LEDs systems (Activity LED , Status LED ) and 3-LEDs systems (Activity LED , Locate LED , Fail LED ).
So can you clarify about led namings on bays like mine?
Activity led - am I right that it is led which blinks when you write/read data from the drive?
Status led - am I right that it is Power led? Or what is status?
Access led - am I right it is the same as Activity led?

> So my bay is 2-led? If not, then why there is no mentioning in man about such bays with one led (Access Led, power LED is not counted it seems). May it be added to man page?

In ledmon we are following IBPI, see:
https://en.wikipedia.org/wiki/International_Blinking_Pattern_Interpretation
Looks like your hardware doesn't. Please double check in documentation.
I think that those leds aren't manageable from system.

> Why it says led management not supported? I have In ledctl -L this output /sys/devices/pci0000:00/0000:00:17.0 (AHCI). So is my controller supported?

This is build-in AHCI controller, please confirm that by checking lspci -s 0000:00:17.0.
You bay is probably transparent in system tree, each drive has direct SATA connection to controller.
It means that controller is supported but bay doesn't. There is no way to communicate with led via SATA cable. So, you are able to impose the state to the controller but you don't see a result on led in bay.

> LED management ( AHCI ) and SAF-TE protocols are not supported.

Yeah, I agree that something is wrong here, we will work to made it clear.

Ashark, 2020-11-23:
Yes, that is internal controller:
```
$ lspci -s 0000:00:17.0
00:17.0 SATA controller: Intel Corporation 200 Series PCH SATA controller [AHCI mode]
```
> In ledmon we are following IBPI

Now I realize it is only for backplanes, but not just sata slots, connected with plain sata cable. And seems there is no way to connect such a backplane on a usual desktop motherboard.

> There is no way to communicate with led via SATA cable.

Thank you for clarification.

> we will work to make it clear.

I think it is just ok to write: `SAF-TE protocols is not supported.`.
I have no other questions for this issue. You can close it if you made an internal note to fix a man page or close it after it is done.

mtkaczyk, 2020-11-24:
Thanks,
I will close it after fix.

Mariusz

PR #77 (2020-12-15): Update ledmon and ledctl man pages
This commit fixes #76
Update the list of supported LED management methods in the man pages
As ledmon currently supports a variety of different platforms and
communication methods, describing all of the combinations in the man
page will be difficult to achieve, so it's an attempt to keep man page
in a general form.
commit ac0b1760 (2020-12-15): Update ledmon and ledctl man pages (#77)

This commit fixes #76
Update the list of supported LED management methods in the man pages
As ledmon currently supports a variety of different platforms and
communication methods, describing all of the combinations in the man
page will be difficult to achieve, so it's an attempt to keep man page
in a general form.
closed 2020-12-15 by mtkaczyk