md-raid-utilities/ledmon #78: pod errors

p-alik, 2020-12-23:
The watchful eye of github caught some pod errors:
* [ledctl.pod](https://github.com/intel/ledmon/blob/master/doc/ledctl.pod#pod-errors)
* [ledmon.pod](https://github.com/intel/ledmon/blob/master/doc/ledmon.pod#pod-errors)

Regarding to `perl-doc` that patch should solve the issue for ledctl.pod`:

```
diff --git a/doc/ledctl.pod b/doc/ledctl.pod
index 7066003..e72d8df 100644
--- a/doc/ledctl.pod
+++ b/doc/ledctl.pod
@@ -42,9 +42,9 @@ storages.
 
 =item * B<SES-2 and SMP> for SAS devices,
 
-=item B<LED messages over SGPIO> for SATA,
+=item * B<LED messages over SGPIO> for SATA,
 
-=item B<VMD and NPEM> for PCIe.
+=item * B<VMD and NPEM> for PCIe.
 
 =back
 
@@ -78,20 +78,20 @@ The ledctl application is a part of Intel(R) Enclosure LED Utilities.
 
 =item * SGPIO (Serial GPIO) - SFF-8485
 
-=item IBPI (International Blinking Pattern Interpretation) - SFF-8489
+=item * IBPI (International Blinking Pattern Interpretation) - SFF-8489
 
-=item LED Enclosure management messages - AHCI specification rev 1.3,
+=item * LED Enclosure management messages - AHCI specification rev 1.3,
 section 12.2.1.
 
-=item SAS (Serial Attached SCSI) - T10/1760-D
+=item * SAS (Serial Attached SCSI) - T10/1760-D
 
-=item SES-2 (SCSI Enclosure Services-2) - T10/1559-D
+=item * SES-2 (SCSI Enclosure Services-2) - T10/1559-D
 
-=item SMP (Serial Management Protocol) - T10/1760-D
+=item * SMP (Serial Management Protocol) - T10/1760-D
 
-=item NPEM (Native PCIe Enclosure Management) - PCIe base specification rev 4.0
+=item * NPEM (Native PCIe Enclosure Management) - PCIe base specification rev 4.0
 
-=item VMD (Intel(R) Volume Management Device) - Intel(R) VROC (VMD NVMe RAID) Quick
+=item * VMD (Intel(R) Volume Management Device) - Intel(R) VROC (VMD NVMe RAID) Quick
 Configuration Guide rev 1.2
 
 =back
```

mtkaczyk, 2021-01-04:
Hello,
Could you create pull request?

Mariusz

p-alik, 2021-01-05:
@mtkaczyk, I did a PR as requested.

closed 2021-01-15 by mtkaczyk