md-raid-utilities/ledmon #81: Turning off the locate LED after ejecting the device

jamesyoungman, 2021-05-09:
I use `ledctl` to identify which disk to pull when a disk fails:    `ledctl locate=/dev/sdn`

This works well.   But, once I eject the faulted device, `ledctl locate_off=dev/sdn` no longer works (since, I suppose, the `/dev` node is no longer there).

Is there a solution or workaround for this?

mtkaczyk, 2021-05-28:
Hello,
Unfortunately, there is no support for blinking empty slots.  We are planning to add it in the future.
I prepared workaround for this problem some time ago.
This will clear every failure led by writing directly to em_message . 
I hope it helps.

``` bash
#!/bin/bash
# This script was designed to remove "failure" led from each empty slot.
# Verified to work with SATA drives.
STATE_FAILURE="400000"
STATE_NORMAL="0"

# locate all possible leds
LEDS=$(find /sys/devices/ -name em_message)

for LED in $LEDS; do
        # verify whether led management is supported
        [ $(cat "$LED"_supported) == "led" ] || continue
        # if state is different than failure then ignore
        [ $(cat $LED) == "$STATE_FAILURE" ] || continue
        HOST=$(echo $LED | awk -F 'scsi_host' '{print $1}')
        # check if disk is present
        [ -n $(ls $HOST | grep target) ] || continue
        echo "Removing failure led for: $HOST"
        echo "$STATE_NORMAL" > $LED
done
```

mtkaczyk, 2021-11-23:
NO, update, closing.

closed 2021-11-23 by mtkaczyk