md-raid-utilities/ledmon #84: Dell R7515 "enclosure management not supported"

Timvrakas, 2021-09-04:
I've installed ledmon 0.95 from source on proxmox buster, on a Dell R7515 (AMD Epyc) with 2x4 SATA bays.

Running: `./src/ledctl --all -L` yeilds

```
ledctl: SGPIO EM not supported for /sys/devices/pci0000:80/0000:80:08.2/0000:84:00.0
ledctl: controller discovery: /sys/devices/pci0000:80/0000:80:08.2/0000:84:00.0 - enclosure management not supported.
ledctl: Unable to determine Dell Server type
ledctl: Couldn't find base EM path for /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0
ledctl: controller discovery: /sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0 - enclosure management not supported.
ledctl: Unable to determine Dell Server type
ledctl: Couldn't find base EM path for /sys/devices/pci0000:40/0000:40:01.1/0000:41:00.0
ledctl: controller discovery: /sys/devices/pci0000:40/0000:40:01.1/0000:41:00.0 - enclosure management not supported.
ledctl: Unable to determine Dell Server type
ledctl: Couldn't find base EM path for /sys/devices/pci0000:80/0000:80:03.1/0000:81:00.0
ledctl: controller discovery: /sys/devices/pci0000:80/0000:80:03.1/0000:81:00.0 - enclosure management not supported.
ledctl: SGPIO EM not supported for /sys/devices/pci0000:80/0000:80:08.3/0000:85:00.0
ledctl: controller discovery: /sys/devices/pci0000:80/0000:80:08.3/0000:85:00.0 - enclosure management not supported.
```

Anyone have any hints for what might be the issue?

mtkaczyk, 2021-09-06:
Hi,
Could you retest with other versions?
Is it a regression?

Thanks,
Mariusz

mtkaczyk, 2021-11-23:
No update, closing.

prithivi17, 2024-09-16:
> I've installed ledmon 0.95 from source on proxmox buster, on a Dell R7515 (AMD Epyc) with 2x4 SATA bays.
>
> Running: `./src/ledctl --all -L` yeilds
>
> ```
> ledctl: SGPIO EM not supported for [...]

Hello actually, im getting the same error message in my dell server, also i can see some strange behaviour that the anonymous memory (heap memory) of the ledmon process getting increased exponentially, only after stopping the ledmon process makes the anonymous memory free. Are you facing any similar memory leak issue? Kindly share your perception about this.

ktanska, 2024-09-18:
Hi @prithivi17 
did you test ledmon 0.95 or the newest 1.0.0 version?

prithivi17, 2024-09-18:
> Hi @prithivi17 did you test ledmon 0.95 or the newest 1.0.0 version?

i have tried the 0.95 version, i can see the anonymous memory growth in 0.95,0.96,0.97, it seems like newest version 1.0.0 is working fine. Can i know why the memory leak issue in the below versions

ktanska, 2024-09-18:
There were a lot of changes between v0.97 and v1.0.0. We added library support there (#115) , and this issue could be resolved with this pull request.
I can't help to find resolving patch, because I don't have hardware to reproduce this. If you can help bisecting, I will really appreciate. Instructions how to compile the code you can find in README.

minorsatellite, 2024-09-18:
I have never gotten ledmon to work on my AMD box, its been a 100% fruitless endeavor. I ended up disabling it.

prithivi17, 2024-09-20:
> I have never gotten ledmon to work on my AMD box, its been a 100% fruitless endeavor. I ended up disabling it.

Can you share how you done it ? have'nt you faced any memory leak and which version have you tried?
@minorsatellite

prithivi17, 2024-09-26:
> There were a lot of changes between v0.97 and v1.0.0. We added library support there (#115) , and this issue could be resolved with this pull request. I can't help to find resolving patch, because I [...]

i have found the block of code in v0.95 & v0.97 where the memory leak is occuring, how can i post the patch for this?

ktanska, 2024-09-26:
Hi @prithivi17 
Please explain more, where would you like to have it fixed? Do you want to have patch added to released package?
You mentioned, that it is fixed in ledmon 1.0.0, maybe the most suitable option is to switch to this release.

prithivi17, 2024-09-30:
> Hi @prithivi17 Please explain more, where would you like to have it fixed? Do you want to have patch added to released package? We don't update released packages. You mentioned, that it is fixed in ledmon 1.0.0, maybe the most suitable option is to switch to this release. Which OS are you using? OS vendor is responsible for updating ledmon package in system or backport patch and share updated package. We work continuously with RedHat and SLES to update this package, but other systems may not have the latest package installed always.

Hi @ktanska , since the memory leak is only occurring on servers with the AMD SATA controller. Btw I have tested it on Debian 11.7 and 12.6 OS, as the repository version of ledmon is still 0.95. I thought it might be helpful to address the memory leak issue in the versions prior to the latest version, i.e., v1.

mtkaczyk, 2024-09-30:
> Hi @ktanska , since the memory leak is only occurring on servers with the AMD SATA controller. Btw I have tested it on Debian 11.7 and 12.6 OS, as the repository version of ledmon is still 0.95. I [...]

This is not in our control. We are upstream developers but we are not owning ledmon in debian. The best you can do is to ask debian maintainer to backport the fix. We are here to help if maintainer needs support with backporting (if backport is not clean).

Fortunately, I know maintainer in debian- @woodrow-shen could you please take a look and consider backporting?

@prithivi17 please link the fix you have found here (maintainer must know which fix to backport). You should also open issue in debian: https://packages.debian.org/sid/ledmon

mtkaczyk, 2024-10-01:
Here a fix:
https://github.com/intel/ledmon/commit/607d51cab9065fb120719ebeb7b6477d0bc45fd9

This is done after reworking the code, so it won't be applicable for ledmon v.097 and must be applied manually.

prithivi17, 2024-10-03:
> Here a fix: [607d51c](https://github.com/intel/ledmon/commit/607d51cab9065fb120719ebeb7b6477d0bc45fd9)
> 
> This is done after reworking the code, so it won't be applicable for ledmon v.097 and must be applied manually.

Hi @mtkaczyk , I understand that you have made significant changes from version 0.97 to 1.0. The proposed fix I have done addresses the memory leak found in one function in version 0.95 and two functions in version 0.97 related to the AMD SATA controller, which I have tested on my end. I kindly request your approval for this change so that I can raise a bug report (memory leak) for ledmon 0.95, the version in their repository.

mtkaczyk, 2024-10-03:
Hello, @prithivi17 
sure, you have my ack. The proposed change looks safe.
Good luck with Debian!

mtkaczyk, 2024-10-04:
Closing.
Thanks everyone!

closed 2024-10-04 by mtkaczyk