md-raid-utilities/ledmon #86: ledctl not working with simple Intel SATA controller

tuxedo0801, 2021-09-29:
Hi there,

I'm running a Debian Testing system on a Biostar HM110MHV3 mainboard and having issues controling the LEDs of my SATA devices, connected to an ICYBOX IB-55SSK (https://icybox.de/en/product.php?id=336 --> as far as I know, just wires SATA from mainboard to the connected disks, nothing special inbetween but a nice case and a fan), which has a combined power/activity LED for each disk. 

Two disks are an mdadm raid1, so I not only have /dev/sdX, but also a /dev/md127
other disks run in single mode

ledctl -L says:

```
# ledctl -L
/sys/devices/pci0000:00/0000:00:17.0 (AHCI)
```
If I set loglevel to all, I get f.i. this:

```
# ledctl locate=/dev/sdb
ledctl: Unable to issue IPMI command GetSystemInfo
ledctl: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sdc, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
```

LED does nothing... tried different pattern. Same ... --> nothing.

I'm not very deep into the underlying technical stuff required for this. Maybe this helps to identify the problem?

```
# lspci | grep -i AHCI
00:17.0 SATA controller: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset SATA Controller [AHCI Mode] (rev 31)
```

What am I doing wrong? Or is my system simply not compatible? Any way of letting drive-led blink to show hw-issues or to identify drive?

br,
Alex

tuxedo0801, 2021-09-30:
btw: as current debian testing has an issue with ledmon (see: https://www.mail-archive.com/debian-bugs-rc@lists.debian.org/msg598861.html), I manually fixed the symlink to the libsgutil that comes with debian... I first thought, that there is an issue wth an non-matching libsgutil... but then tried to compile ledmon from source with latest changes from this github repo... same effect.

So.... still trying to get an idea of whats wrong with my doing or setup.

apaszkie, 2021-09-30:
Hi Alex,

I'm sorry but your backplane does not support this. The datasheet and manuals mention only Power and Access LEDs, not Status LED. It does not have a [SGPIO](https://en.wikipedia.org/wiki/SGPIO) connector, so there is no way to transmit LED status data - SATA cables are not used for those signals.

Regards,
Artur

tuxedo0801, 2021-09-30:
Hmm, okay, I thought it's possible for (at least) Intel bases SATA Controller to influence the typical activity LED. But obviously, this is not possible. Too bad.

But okay, then I need to find other ways and concepts to solve my actual problem of drive identifying. 

Thanks @apaszkie  for clarification.

closed 2021-09-30 by tuxedo0801