md-raid-utilities/ledmon #102: 0.96: build fails

kloczek, 2022-06-01:
Looks like linking fails with `-Werror=format-truncation`
<details>

```console
+ cd ledmon-0.96
+ autoreconf -fiv
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: /usr/bin/autoconf --force
configure.ac:14: warning: The macro `AC_PROG_CC_C99' is obsolete.
configure.ac:14: You should run autoupdate.
./lib/autoconf/c.m4:1659: AC_PROG_CC_C99 is expanded from...
configure.ac:14: the top level
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:13: installing './compile'
configure.ac:63: installing './config.guess'
configure.ac:63: installing './config.sub'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
src/Makefile.am: installing './depcomp'
autoreconf: Leaving directory '.'
+ CFLAGS='-O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os'
+ CXXFLAGS='-O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none'
+ FFLAGS='-O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -I/usr/lib64/gfortran/modules'
+ FCFLAGS='-O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -I/usr/lib64/gfortran/modules'
+ LDFLAGS='-Wl,-z,relro -Wl,--as-needed -Wl,--gc-sections -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin -Wl,--build-id=sha1'
+ CC=/usr/bin/gcc
+ CXX=/usr/bin/g++
+ FC=/usr/bin/gfortran
+ AR=/usr/bin/gcc-ar
+ NM=/usr/bin/gcc-nm
+ RANLIB=/usr/bin/gcc-ranlib
+ export CFLAGS CXXFLAGS FFLAGS FCFLAGS LDFLAGS CC CXX FC AR NM RANLIB
+ ./configure --build=x86_64-redhat-linux-gnu --host=x86_64-redhat-linux-gnu --program-prefix= --disable-dependency-tracking --disable-silent-rules --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --datadir=/usr/share --includedir=/usr/include --libdir=/usr/lib64 --libexecdir=/usr/libexec --localstatedir=/var --runstatedir=/run --sharedstatedir=/var/lib --mandir=/usr/share/man --infodir=/usr/share/info
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... yes
checking for make... yes
checking whether make supports the include directive... yes (GNU style)
checking for x86_64-redhat-linux-gnu-gcc... /usr/bin/gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /usr/bin/gcc accepts -g... yes
checking for /usr/bin/gcc option to enable C11 features... none needed
checking whether /usr/bin/gcc understands -c and -o together... yes
checking dependency style of /usr/bin/gcc... none
checking whether C compiler accepts -Wformat -Werror=format-security... yes
checking whether C compiler accepts -Werror=format-overflow=2... yes
checking whether C compiler accepts -Werror=format-truncation=1... yes
checking whether C compiler accepts -Werror=shift-negative-value... yes
checking whether C compiler accepts -Werror=alloca... yes
checking whether C compiler accepts -Werror=missing-field-initializers... yes
checking whether C compiler accepts -Werror=format-signedness... yes
checking whether make supports nested variables... (cached) yes
checking for x86_64-redhat-linux-gnu-pkg-config... no
[... 294 lines cut ...]
    inlined from 'block_device_init' at block.c:239:11:
/usr/include/bits/stdio2.h:71:10: note: '__builtin___snprintf_chk' output 11 or more bytes (assuming 4106) into a destination of size 4096
   71 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^
In function '_is_smp_cntrl',
    inlined from '_get_type' at cntrl.c:238:8,
    inlined from 'cntrl_device_init' at cntrl.c:398:9,
    inlined from '_cntrl_add' at sysfs.c:334:32,
    inlined from '_check_cntrl' at sysfs.c:376:3:
cntrl.c:184:41: error: '/bsg/sas_' directive output may be truncated writing 9 bytes into a region of size between 0 and 4095 [-Werror=format-truncation=]
  184 |                                         "%s/%s/bsg/sas_%s", path, p, p);
      |                                         ^
In function 'snprintf',
    inlined from '_is_smp_cntrl' at cntrl.c:183:5,
    inlined from '_get_type' at cntrl.c:238:8,
    inlined from 'cntrl_device_init' at cntrl.c:398:9,
    inlined from '_cntrl_add' at sysfs.c:334:32,
    inlined from '_check_cntrl' at sysfs.c:376:3:
/usr/include/bits/stdio2.h:71:10: note: '__builtin___snprintf_chk' output 11 or more bytes (assuming 4106) into a destination of size 4096
   71 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/gcc returned 1 exit status
```
</details>

ktanska, 2022-06-01:
Could you provide me versions of gcc and make packages?

ktanska, 2022-06-06:
Thanks. In attached log I see that these errors are expected and coms from additional flag "-Werror=format-truncation". We use buffer with PATH_MAX size, which is the same as components size. In case that all buffer will be written, output will be truncated, which is expected.
I suggest you to turn off this flag to avoid these errors.

kloczek, 2022-06-06:
Please do not swipe that under the carpet.

mtkaczyk, 2022-06-06:
Linux supports name with a size up to 4096, which is PATH_MAX. That size shall be our max length.
If we are concatenating many paths like in those cases, we accept the risk of buffer truncation. I believe that issues cannot occur real live, especially that we are scanning sysfs, interface created by kernel.
I don't see any advantage of creating dedicated and precisely calculated memory for each buffer. For me it is more bug prone and makes code less understandable.

For me it is not an issue but if it is important to you we will try to meet your requirements. Please elaborate why this flag is important for you.

mtkaczyk, 2022-06-14:
No feedback, closing.
Fell free to reopen the issue if necessary.

kloczek, 2022-06-14:
> Please do not swipe that under the carpet.

I've provided feedback https://github.com/intel/ledmon/issues/102#issuecomment-1147223050

kloczek, 2022-06-14:
Simple I don't see any reason why only ledmon shoud be only exception about not pasing compile with
`-Werror=format-truncation`. I have now cleaned +4.2k packages. All are build with  `-Werror=format-truncation`.

mtkaczyk, 2022-06-14:
Sure, that is a good point- it is a common standard now.
To fix that we need to:
- add verification to our CI.
- resolve existing issues.

It is not critical, so I will create appropriative internal task, hopefully we can resolve that until next release.
Reopening then.

closed 2023-08-28 by mtkaczyk