md-raid-utilities/ledmon #107: ledctl: unrecognized option '--amd_sgpio3'

pickaaaaa, 2022-08-26:
On redhat8.4,centos8.4 and ubuntu 20.04, I have tried the ledmon tool of version 0.93 and it works.
On SLES 15SP3,I have tried the ledmon tool from version 0.93 to 0.96 with the following logs.

localhost:~ # ledctl --amd_sgpio3 locate=/dev/sda
ledctl: unrecognized option '--amd_sgpio3'
localhost:~ # systemctl start ledmon.service
localhost:~ # systemctl status ledmon.service
● ledmon.service - Enclosure LED Utilities
     Loaded: loaded (/usr/lib/systemd/system/ledmon.service; disabled; vendor preset: disabled)
     Active: active (running) since Thu 2022-08-25 20:42:03 CST; 5s ago
   Main PID: 6017 (ledmon)
      Tasks: 1
     CGroup: /system.slice/ledmon.service
             └─6017 /usr/sbin/ledmon --foreground

Aug 25 20:42:04 localhost ledmon[6017]: ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: ledmon[6017]: Unsupported AMD interface
Aug 25 20:42:04 localhost ledmon[6017]: Unsupported AMD interface
localhost:~ # ledctl --amd_sgpio3 locate=/dev/sda
ledctl: unrecognized option '--amd_sgpio3'
localhost:~/amdledmon/ledmon-0.95/src # ledctl --all -L
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
ledctl: IPMI Error: 80
ledctl: Unable to determine Dell Server type
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
ledctl: Unsupported AMD interface
/sys/devices/pci0000:99/0000:99:08.2/0000:b1:00.0 (AMD)
/sys/devices/pci0000:c3/0000:c3:08.3/0000:c9:00.0 (AMD)
/sys/devices/pci0000:46/0000:46:08.3/0000:4c:00.0 (AMD)
/sys/devices/pci0000:c3/0000:c3:08.2/0000:c8:00.0 (AMD)
/sys/devices/pci0000:46/0000:46:08.2/0000:4b:00.0 (AMD)
/sys/devices/pci0000:12/0000:12:08.3/0000:2b:00.0 (AMD)
/sys/devices/pci0000:12/0000:12:08.2/0000:2a:00.0 (AMD)
/sys/devices/pci0000:99/0000:99:08.3/0000:b2:00.0 (AMD)

mtkaczyk, 2022-08-29:
Hello,
ledctl is not connected with ledmon. This is simple one task app to set requested state.
So far I know, there is no "--amd_sgpio3" parameter, ledmon chooses controller automatically.
If there is a regression, then it is probably caused by changes to more strict parameters policy.

could you please run:
> ledctl --locate=/dev/sda -x --all

Please also describe more detailed what is the problem. what does not work now?

Thanks,
Mariusz

mtkaczyk, 2022-09-26:
No feedback, closing issue.

closed 2022-09-26 by mtkaczyk