md-raid-utilities/ledmon #121: ledctl slot commands "--controller" option is misleading

apaszkie, 2023-02-02:
Based on ledctl --help it is not clear how to use the slot commands ("--list-slots", "--get-slot", "--set-slot") correctly:

```
--list-controllers                                                    -L                                      Displays list of controllers detected by ledmon.
--list-slots --controller CONTROLLER                                  -P -c CONTROLLER                        List slots under the controller, their led states, slot numbers and devnodes connected.
--get-slot --controller CONTROLLER --device DEVNODE / --slot SLOT     -G -c CONTROLLER -d DEVNODE / -p SLOT   Prints slot information, its led state, slot number and devnode.
--set-slot --controller CONTROLLER --slot SLOT --state STATE          -S -c CONTROLLER -p SLOT -s STATE       Sets given state for chosen slot under the controller.
```

It is not clear that the "--controller" parameter actually requires the controller **type**, e.g. "VMD", not the actual controller path printed by --list-controllers. The ledctl manpage does mention this for the "--list-slots" command, but not for "--get-slot" and "--set-slot".

ktanska, 2023-02-13:
Internal task submitted. Will be resolved with #123 .

ktanska, 2023-05-16:
Fixed in upstream. Parameter renamed to "--controller_type".

apaszkie, 2023-05-16:
Great. Thank you. You can close the issue.

closed 2023-05-16 by ktanska