md-raid-utilities/ledmon #123: Fix --help for ledctl

mtkaczyk, 2023-02-13:
The column size should be limited to 100 chars (compatibility with terminals and kernel style).

Help should be divided into
- general_help;
print modes and IBPI (backward compatibility)
- help per mode 
print mode specific parameters if `--help` is used after mode

bkucman, 2023-08-31:
Needs to be done after https://github.com/intel/ledmon/pull/151

yihaoshih, 2023-09-04:
I've confirmed the issue. Can someone please assist me? Thank you.

root@ppp:~/ledmon-master/doc# ledctl locate=/dev/sda
/dev/shm/ledmon.conf: does not exist, using global config file
/etc/ledmon.conf: does not exist, using built-in defaults
ledctl: /dev/sda: device not supported
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.
root@ppp:~/ledmon-master/doc# ledctl -L
/dev/shm/ledmon.conf: does not exist, using global config file
/etc/ledmon.conf: does not exist, using built-in defaults
/sys/devices/pci0000:00/0000:00:0e.0/pci10000:e0/10000:e0:17.0 (AHCI)
/sys/devices/pci0000:00/0000:00:0e.0 (VMD)