md-raid-utilities/ledmon #129: Ledmon changes device name assigned after every loop.

mtkaczyk, 2023-03-22:
If there are multiple namespaces defined, I can see that ledmon is changing namespace assigned in every loop:
> ll /sys/block/| grep nvme7
lrwxrwxrwx  1 root root 0 Mar 21 19:35 nvme7n1 -> ../devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n1
lrwxrwxrwx  1 root root 0 Mar 21 19:35 nvme7n2 -> ../devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n2

> ledmon[56099]: NAME CHANGED /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n1 to /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n2
ledmon[56099]: NAME CHANGED /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n2 to /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n1

Next loop:
> ledmon[56099]: NAME CHANGED /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n1 to /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n2
ledmon[56099]: NAME CHANGED /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n2 to /sys/devices/pci0000:5d/0000:5d:05.5/pci10001:00/10001:00:03.0/10001:02:00.0/nvme/nvme7/nvme7n1

It could affect scenarios with multiple namespaces defined and VROC RAID array on top of first one, however I don't dig into it. Need to be investigated.

ktanska, 2023-12-01:
Probably fix is needed in block_compare() function for NPEM.

mtkaczyk, 2026-11-03:
Fixed with #275, thanks @tasleson

closed 2026-11-03 by mtkaczyk