md-raid-utilities/ledmon #178: FIX possible memflood

allanlaal, 2023-09-14:
after 35 days up, ledmon takes too much RAM:

	Private  +   Shared  =  RAM used	Program
	3.6 GiB +  10.5 KiB =   3.6 GiB	ledmon

peluse, 2023-09-14:
might be good to consider adding a quick https://valgrind.org/ test for each PR

tasleson, 2023-09-19:
> after 35 days up, ledmon takes too much RAM:

Are you using a packaged version (provide what version), or building from source repository (provide what commit)?

Thanks

mtkaczyk, 2023-09-25:
@allanlaal,
please provide more details as @tasleson requested. We need to know if it is a recent problem introduced with library.
Then, we can start investigation.

allanlaal, 2023-09-28:
it was in ubuntu 23.04 before version 0.95-2
sadly couldn't find what that previous version was
I'll reopen when I encounter it again

closed 2023-09-28 by allanlaal

mtkaczyk, 2023-09-29:
Thanks for feedback!
The one option I'm considering here is to add valgrind to CI.
Unfortunately, we don't have yet ledmon tests in CI so it must wait.

Thanks,
Mariusz