md-raid-utilities/ledmon #204: Cannot locate multipathing nvme block device by ledctl

jackeichen, 2024-01-31:
When enable multipathing nvme device and Intel VROC technology, the bellow command will failed:
```
# ledctl locate=/dev/nvme1n1
ledctl: /dev/nvme1n1: device not supported
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.

# ledctl locate=/dev/nvme1c1n1
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.
```
The tool version is 0.97, here is my configuration:
```
# ledctl -L
/sys/devices/pci0000:00/0000:00:17.0 (AHCI)
/sys/devices/pci0000:00/0000:00:11.5 (AHCI)
/sys/devices/pci0000:c2/0000:c2:00.5 (VMD)
/sys/devices/pci0000:89/0000:89:02.0/0000:8a:00.0 (SCSI)

# ledctl -P -c VMD
slot: 66              led state: NORMAL          device: /dev/nvme1c1n1
slot: 64              led state: NORMAL          device: (empty)
slot: 67              led state: NORMAL          device: (empty)
slot: 65              led state: NORMAL          device: /dev/nvme0n1

# find / -name nvme0n1
/dev/nvme0n1
/sys/kernel/debug/block/nvme0n1
/sys/class/block/nvme0n1
/sys/devices/pci0000:c2/0000:c2:00.5/pci10000:00/10000:00:02.0/10000:01:00.0/nvme/nvme0/nvme0n1
/sys/devices/virtual/block/md127/slaves/nvme0n1

# find / -name nvme1n1
/dev/nvme1n1
/sys/kernel/debug/block/nvme1n1
/sys/class/block/nvme1n1
/sys/devices/virtual/nvme-subsystem/nvme-subsys1/nvme1n1
/sys/block/nvme1n1
```

mtkaczyk, 2024-02-01:
Hey @jackeichen,
Support of nvme multipath devices has been introduced recently:
https://github.com/intel/ledmon/commit/82c3675cadd1c95058b115c5301683f41372fc72

Please test with latest upstream ledmon to see if it is working now.

jackeichen, 2024-02-04:
It works in my configuration.

Thanks for the reply.

closed 2024-02-04 by jackeichen