md-raid-utilities/ledmon #214: 1.0.0: build fails

kloczek, 2024-03-27:
### Description

Looks like 1.0.0 is not finished and build fails because missing `#include` lines.

### Steps to reproduce bug

clone ->  `autoreconf -fiv`` -> configure  -> make

### Expected behavior

Build should not fail.

### Actual behavior

<details>

```console
[tkloczko@pers-jacek ledmon-1.0.0]$ make -k V=1
make  all-recursive
make[1]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0'
Making all in src
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src'
Making all in common
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/common'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/common'
Making all in lib
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib'
Making all in include
make[4]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib/include'
Making all in led
make[5]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib/include/led'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib/include/led'
make[5]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib/include'
make[5]: Nothing to be done for 'all-am'.
make[5]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib/include'
make[4]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib/include'
make[4]: Entering directory '/home/tkloczko/rpmbuild/BUILD/ledmon-1.0.0/src/lib'
/bin/sh ../../libtool  --tag=CC   --mode=compile /usr/bin/gcc -DHAVE_CONFIG_H -I. -I../..    -I../../src/lib/include -I../../src -I../../config    -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -c -o libledinternal_la-block.lo `test -f 'block.c' || echo './'`block.c
libtool: compile:  /usr/bin/gcc -DHAVE_CONFIG_H -I. -I../.. -I../../src/lib/include -I../../src -I../../config -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -c block.c  -fPIC -DPIC -o .libs/libledinternal_la-block.o
block.c: In function 'block_set_devnode':
block.c:244:22: error: implicit declaration of function 'basename'; did you mean 'rename'? [-Wimplicit-function-declaration]
  244 |         char *name = basename(device->sysfs_path);
      |                      ^~~~~~~~
      |                      rename
block.c:244:22: error: initialization of 'char *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
make[4]: *** [Makefile:630: libledinternal_la-block.lo] Error 1
/bin/sh ../../libtool  --tag=CC   --mode=compile /usr/bin/gcc -DHAVE_CONFIG_H -I. -I../..    -I../../src/lib/include -I../../src -I../../config    -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -c -o libledinternal_la-vmdssd.lo `test -f 'vmdssd.c' || echo './'`vmdssd.c
libtool: compile:  /usr/bin/gcc -DHAVE_CONFIG_H -I. -I../.. -I../../src/lib/include -I../../src -I../../config -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -g -grecord-gcc-switches -pipe -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fdata-sections -ffunction-sections -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -flto=auto -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -Wall -Werror=format-security -Os -c vmdssd.c  -fPIC -DPIC -o .libs/libledinternal_la-vmdssd.o
vmdssd.c: In function 'vmdssd_get_domain':
vmdssd.c:85:29: error: implicit declaration of function 'basename'; did you mean 'rename'? [-Wimplicit-function-declaration]
   85 |                  SYSFS_VMD, basename(path));
      |                             ^~~~~~~~
      |                             rename
vmdssd.c:84:46: warning: format '%s' expects argument of type 'char *', but argument 5 has type 'int' [-Wformat=]
   84 |         snprintf(domain_path, PATH_MAX, "%s/%s/domain",
      |                                             ~^
      |                                              |
      |                                              char *
      |                                             %d
   85 |                  SYSFS_VMD, basename(path));
      |                             ~~~~~~~~~~~~~~
      |                             |
      |                             int
vmdssd.c:89:23: error: passing argument 1 of 'strtok' makes pointer from integer without a cast [-Wint-conversion]
   89 |         return strtok(basename(real_domain_path), ":");
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       |
      |                       int
[... 96 lines cut ...]
</details>

### Environment

OS: Linux x86/64

### Ledmon version

1.0.0

### Ledmon logs

N/A

### Ledctl logs

N/A

### Ledmon supported controllers

N/.A

### Additional information

N/A

kloczek, 2024-03-28:
If you are using rpm based distribution could you please try execute `rpm -ql glibc-headers-x86 | xargs grep basename`?
On my system it shows
```console
[tkloczko@pers-jacek SPECS]$ rpm -ql glibc-headers-x86 | xargs grep basename
grep: /usr/include/arpa: Is a directory
grep: /usr/include/bits: Is a directory
grep: /usr/include/bits/platform: Is a directory
grep: /usr/include/bits/types: Is a directory
grep: /usr/include/finclude: Is a directory
grep: /usr/include/gnu: Is a directory
/usr/include/libgen.h:extern char *__xpg_basename (char *__path) __THROW;
/usr/include/libgen.h:#define basename  __xpg_basename
grep: /usr/include/net: Is a directory
grep: /usr/include/netash: Is a directory
grep: /usr/include/netatalk: Is a directory
grep: /usr/include/netax25: Is a directory
grep: /usr/include/neteconet: Is a directory
grep: /usr/include/netinet: Is a directory
grep: /usr/include/netipx: Is a directory
grep: /usr/include/netiucv: Is a directory
grep: /usr/include/netpacket: Is a directory
grep: /usr/include/netrom: Is a directory
grep: /usr/include/netrose: Is a directory
grep: /usr/include/nfs: Is a directory
grep: /usr/include/protocols: Is a directory
grep: /usr/include/rpc: Is a directory
grep: /usr/include/scsi: Is a directory
/usr/include/string.h:# ifndef basename
/usr/include/string.h:   declare the function if the `basename' macro is available (defined
/usr/include/string.h:extern "C++" char *basename (char *__filename)
/usr/include/string.h:     __THROW __asm ("basename") __nonnull ((1));
/usr/include/string.h:extern "C++" const char *basename (const char *__filename)
/usr/include/string.h:     __THROW __asm ("basename") __nonnull ((1));
/usr/include/string.h:extern char *basename (const char *__filename) __THROW __nonnull ((1));
grep: /usr/include/sys: Is a directory
grep: /usr/include/sys/platform: Is a directory
```

mtkaczyk, 2024-04-02:
FYI, I received similar suggestion in mdadm to add `libgen.h`, so please review my comment:
https://patchwork.kernel.org/project/linux-raid/patch/20240325061537.275811-1-raj.khem@gmail.com/

I think that it applies here. I didn't compare those two basename implementations to determine if we can use POSIX version safely.

On both RHEL-9.2 and Opensuse 15.sp4 I don't have this package:
```
[gklab-localhost ~]# rpm -qa | grep glibc
glibc-gconv-extra-2.34-83.el9_3.7.x86_64
glibc-2.34-83.el9_3.7.x86_64
glibc-common-2.34-83.el9_3.7.x86_64
glibc-langpack-en-2.34-83.el9_3.7.x86_64
glibc-doc-2.34-83.el9_3.7.noarch
glibc-headers-2.34-83.el9_3.7.x86_64
glibc-devel-2.34-83.el9_3.7.x86_6
```

Which distribution you are using?

mtkaczyk, 2024-05-14:
No feedback. Closing.

closed 2024-05-14 by mtkaczyk