md-raid-utilities/ledmon #218: 1.0 configure fails

chowbok, 2024-04-06:
### Description

./configure fails on my system with the following errors:

```
checking whether C compiler accepts -fPIE... yes
./configure: line 13881: syntax error near unexpected token `-pie,'
./configure: line 13881: `AX_CHECK_LINK_FLAG(-pie, AM_LDFLAGS="$AM_LDFLAGS -pie")'
```

### Steps to reproduce bug

./autogen.sh && ./configure

### Expected behavior

Not to crash.

### Actual behavior

See above

### Environment

Arch Linux

### Ledmon version

Attempting to build 1.0

### Ledmon supported controllers

Can't run, won't compile.

mtkaczyk, 2024-04-08:
Hello @chowbok 
Could you please provide automake version you are using?

We are using ubuntu in our CI and we are not experiencing configure issues, for example:
https://github.com/intel/ledmon/actions/runs/8251434163

Could you please test with different automake versions?

mtkaczyk, 2024-04-08:
@chowbok I would like to encourage you to create pull request with arch linux compilation. With that, we will be able to to avoid such issues.

bkucman, 2024-04-08:
Hi @chowbok 
You probably don't have the required package _autoconf-archive_ installed [Package docs](https://www.gnu.org/software/autoconf-archive/)
This package contains the _AX_CHECK_LINK_FLAG_ macro on which the reported error occurred.

If further errors occur, please make sure you have the required packages installed: [Dependencies](https://github.com/intel/ledmon?tab=readme-ov-file#1-dependencies).

chowbok, 2024-04-08:
Thanks! That was the issue. I've notified the Arch package maintainer.

closed 2024-04-08 by chowbok