md-raid-utilities/ledmon #224: ledctl: /dev/nvme0n1: device not supported

chs666zmz, 2024-04-25:
### Description

Hi,

I also had the same issue:
/dev/shm/ledmon.conf: does not exist, using global config file
/etc/ledmon.conf: does not exist, using built-in defaults
ledctl: /dev/nvme2n1: device not supported
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.

And here is I see after running:

lrwxrwxrwx. 1 root root 0 Apr 24 21:47 nvme2n1 -> ../devices/pci0000:37/0000:37:00.5/pci10003:80/10003:80:05.0/10003:8c:00.0/nvme/nvme2/nvme2n1

I looks like that this is different with https://github.com/intel/ledmon/issues/19

### Steps to reproduce bug

1. run: ledctl locate=/dev/nvme2n1
2. got the pop-up:
/dev/shm/ledmon.conf: does not exist, using global config file
/etc/ledmon.conf: does not exist, using built-in defaults
ledctl: /dev/nvme2n1: device not supported
ledctl: IBPI LOCATE: missing block device(s)... pattern ignored.

### Expected behavior

LED should be light-up.

### Actual behavior

Disk LED is not light-up.

### Environment

OS: RH8.2
Controller Type(supported: https://github.com/intel/ledmon/blob/master/doc/ledmon.pod):VMD enabled
Disks($ ls -l /sys/block):
NVMe multipath enabled(Yes|No|N/A)($ cat /sys/module/nvme_core/parameters/multipath)?:

### Ledmon version

0.95

### Ledctl logs

Apr 19 04:46:22   ERROR: /dev/nvme0n1: device not supported
Apr 19 04:46:22 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 19 04:47:41 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 19 04:51:01   ERROR: /dev/nvme0n1: device not supported
Apr 19 04:51:01 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 22 02:08:27   ERROR: /dev/nvme0n1: device not supported
Apr 22 02:08:27 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 22 02:09:27   ERROR: /dev/nvme0n1: device not supported
Apr 22 02:09:27 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 22 02:13:51   ERROR: /dev/nvme0n1: device not supported
Apr 22 02:13:51 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 22 02:14:56   ERROR: /dev/nvme0n1: device not supported
Apr 22 02:14:56 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 22 02:15:18   ERROR: /dev/nvme10n1: device not supported
Apr 22 02:15:18 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 02:22:45   ERROR: /dev/nvme0n1: device not supported
Apr 23 02:22:45 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 02:22:54   ERROR: /dev/nvme0n1: device not supported
Apr 23 02:22:54 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 02:23:54   ERROR: /dev/nvme0n1: device not supported
Apr 23 02:23:54 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 02:24:42   ERROR: /dev/nvme0n1: device not supported
Apr 23 02:24:42 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 02:26:03   ERROR: /dev/nvme0n1: device not supported
Apr 23 02:26:03   ERROR: bs - unknown pattern name.
Apr 23 02:26:13   ERROR: /dev/nvme0n1: device not supported
Apr 23 02:26:13 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 02:26:59   ERROR: /dev/nvme1n1: device not supported
Apr 23 02:26:59 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 03:18:30   ERROR: failure_off - unknown pattern name.
Apr 23 23:08:31   ERROR: /dev/nvme0n1: device not supported
Apr 23 23:08:31 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 23:10:51   ERROR: /dev/nvme0n1: device not supported
Apr 23 23:10:51 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 23 23:19:47 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 00:16:46   ERROR: /dev/nvme0n1: device not supported
Apr 24 00:16:46 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 00:17:20   ERROR: /dev/nvme0n1: device not supported
Apr 24 00:17:20 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 02:16:53 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 02:17:01 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 02:17:04 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 02:18:50   ERROR: /dev/nvme0n1: device not supported
Apr 24 02:18:50 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 02:48:29   ERROR: /dev/nvme16n1: device not supported
Apr 24 02:48:29 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:15:23   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:15:23 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:15:33   ERROR: /dev/nvme23n1: device not supported
Apr 24 03:15:33 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:37:52   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:37:52 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:38:30   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:38:30 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:38:34   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:38:34 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:38:35   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:38:35 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:38:37   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:38:37 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:39:23   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:39:23 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:39:24   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:39:24 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:39:26   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:39:26 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:46:30   ERROR: /dev/nvme22n1: device not supported
Apr 24 03:46:30 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:20   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:20 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:22   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:22 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:24   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:24 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:25   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:25 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:26   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:26 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:27   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:27 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:28   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:28 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:48:29   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:48:29 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 03:57:33   ERROR: /dev/nvme21n1: device not supported
Apr 24 03:57:33 WARNING: IBPI FAILURE: missing block device(s)... pattern ignored.
Apr 24 21:48:58   ERROR: /dev/nvme2n1: device not supported
Apr 24 21:48:58 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 21:52:03   ERROR: /dev/nvme2n1: device not supported
Apr 24 21:52:03 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.
Apr 24 21:55:00   ERROR: /dev/nvme2n1: device not supported
Apr 24 21:55:00 WARNING: IBPI LOCATE: missing block device(s)... pattern ignored.

### Ledmon supported controllers

/sys/devices/pci0000:00/0000:00:17.0 (AHCI)
/sys/devices/pci0000:a7/0000:a7:00.5 (VMD)
/sys/devices/pci0000:97/0000:97:00.5 (VMD)
/sys/devices/pci0000:15/0000:15:00.5 (VMD)
/sys/devices/pci0000:26/0000:26:00.5 (VMD)
/sys/devices/pci0000:b7/0000:b7:00.5 (VMD)
/sys/devices/pci0000:37/0000:37:00.5 (VMD)
/sys/devices/pci0000:48/0000:48:00.5 (VMD)
/sys/devices/pci0000:c7/0000:c7:00.5 (VMD)
/sys/devices/pci0000:00/0000:00:18.0 (AHCI)
/sys/devices/pci0000:59/0000:59:00.5 (VMD)
/sys/devices/pci0000:00/0000:00:00.5 (VMD)
/sys/devices/pci0000:d7/0000:d7:00.5 (VMD)
/sys/devices/pci0000:80/0000:80:00.5 (VMD)

chs666zmz, 2024-04-25:
This is what I got after running "ll /sys/block | grep nvme":
lrwxrwxrwx. 1 root root 0 Apr 24 21:47 nvme2n1 -> ../devices/pci0000:37/0000:37:00.5/pci10003:80/10003:80:05.0/10003:8c:00.0/nvme/nvme2/nvme2n1

lrwxrwxrwx. 1 root root 0 Apr 24 21:47 nvme0n1 -> ../devices/pci0000:37/0000:37:00.5/pci10003:80/10003:80:01.0/10003:8a:00.0/nvme/nvme0/nvme0n1

It looks like that they are in the support list.

bkucman, 2024-04-25:
Hi @chs666zmz 
Thanks for reporting the issue. 

As a first step, please test whether the issue also appears on the latest release _v1.0.0_ , ledmon 0.95 is quite old.
You can do this by cloning the repository and switching to tag _v1.0.0_ or by downloading source code archive of release [Releases](https://github.com/intel/ledmon/releases)

Necessary packages and building instructions are available in the [README](https://github.com/intel/ledmon/blob/main/README.md)

In short, please run commands listed below
- Run autogen.sh to generate compiling configurations: `./autogen.sh`
- Run `./configure`
- Run `make`
Note: Errors during build may mean that the dependencies described in the README are missing.

Built `ledmon` does not have to be installed, you can test it by running the ledctl binary `src/ledctl/ledctl` with your arguments. 

Let us know if the problem persists, if so, run the additional commands using `v1.0.0` ledmon and collect output and logs.
- src/ledctl/ledctl --list-slots -n VMD --log-level=all
- src/ledctl/ledctl --list-slots -n  NPEM --log-level=all
- src/ledctl/ledctl --list-controllers --log-level=all

mtkaczyk, 2024-05-28:
No feedback from submitter. Closing.

closed 2024-05-28 by mtkaczyk