md-raid-utilities/ledmon #236: [QUESTION] LED indication not working on Supermicro AMD system

CendioMartin, 2024-06-20:
### Question

I'm having some issues getting led location to turn on. I don't find anything obviously wrong when executing 'ledctl', nor in the logs. 

For example, `ledctl locate=/dev/sdb`  produces no output, which I assume means that it completed successfully, however the LED on the disk bay does not light up at all. 

I've tried to locate on all disks.

Is this expected to work?

### Environment

Server is Supermicro Model AS-1114S-WTRT, with a Supermicro H12SSW-NT motherboard, and a AMD EPYC 7313P processor. 
Running RHEL 8.10 with ledmon version 0.97 from RHEL repo.

[root@storage ~]#  lspci -vvv   | grep -i sata
43:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 51) (prog-if 01 [AHCI 1.0])
	Capabilities: [d0] SATA HBA v1.0 InCfgSpace
44:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 51) (prog-if 01 [AHCI 1.0])
	Capabilities: [d0] SATA HBA v1.0 InCfgSpace
c5:00.0 SATA controller: ASMedia Technology Inc. ASM1061/ASM1062 Serial ATA Controller (rev 02) (prog-if 01 [AHCI 1.0])
	DeviceName: ASMedia SATA 6G
	
[root@storage]# ledctl --all -L
ledctl: AMD Drive: port 6, ata port 8, drive bay 2, initiator 1
ledctl: AMD Drive: port 7, ata port 17, drive bay 1, initiator 1
ledctl: (raid_device_init) path: md0, level=5, state=6, degraded=0, disks=5, type=1
ledctl: (raid_device_init) path: md1, level=5, state=7, degraded=0, disks=3, type=1
ledctl: (_set_block_state): device: sdf, state: NORMAL
ledctl: (_set_block_state): device: sde, state: NORMAL
ledctl: (_set_block_state): device: sdd, state: NORMAL
ledctl: (_set_block_state): device: sdc, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sdi, state: NORMAL
ledctl: (_set_block_state): device: sdh, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
/sys/devices/pci0000:40/0000:40:08.2/0000:43:00.0 (AMD)
/sys/devices/pci0000:40/0000:40:08.3/0000:44:00.0 (AMD)

[root@storage]# ls -al /dev/disk/by-path/
total 0
drwxr-xr-x. 2 root root 480 May 28 11:45 .
drwxr-xr-x. 7 root root 140 May 28 11:45 ..
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-1 -> ../../sdb
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-1-part1 -> ../../sdb1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-2 -> ../../sdc
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-2-part1 -> ../../sdc1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-3 -> ../../sdd
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-3-part1 -> ../../sdd1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-4 -> ../../sde
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-4-part1 -> ../../sde1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-5 -> ../../sdf
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-5-part1 -> ../../sdf1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-6 -> ../../sdg
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-6-part1 -> ../../sdg1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-7 -> ../../sdh
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-7-part1 -> ../../sdh1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:43:00.0-ata-8 -> ../../sdi
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:43:00.0-ata-8-part1 -> ../../sdi1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:44:00.0-ata-1 -> ../../sdj
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:44:00.0-ata-1-part1 -> ../../sdj1
lrwxrwxrwx. 1 root root   9 May 28 11:45 pci-0000:c5:00.0-ata-1 -> ../../sda
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:c5:00.0-ata-1-part1 -> ../../sda1
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:c5:00.0-ata-1-part2 -> ../../sda2
lrwxrwxrwx. 1 root root  10 May 28 11:45 pci-0000:c5:00.0-ata-1-part3 -> ../../sda3

Output of `ledctl localte=/dev/sdb --all`:

```
ledctl: AMD Drive: port 6, ata port 8, drive bay 2, initiator 1
ledctl: AMD Drive: port 7, ata port 17, drive bay 1, initiator 1
ledctl: (raid_device_init) path: md0, level=5, state=7, degraded=0, disks=5, type=1
ledctl: (raid_device_init) path: md1, level=5, state=6, degraded=0, disks=3, type=1
ledctl: (_set_block_state): device: sdf, state: NORMAL
ledctl: (_set_block_state): device: sde, state: NORMAL
ledctl: (_set_block_state): device: sdd, state: NORMAL
ledctl: (_set_block_state): device: sdc, state: NORMAL
ledctl: (_set_block_state): device: sdb, state: NORMAL
ledctl: (_set_block_state): device: sdi, state: NORMAL
ledctl: (_set_block_state): device: sdh, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
ledctl: 
ledctl: Setting LOCATE_OFF...
ledctl: 	device: .../ata7/host6/target6:0:0/6:0:0:0/block/sdf
ledctl: 	buffer: .../ata8/host7/scsi_host/host7/em_buffer
ledctl: AMD Drive: port 5, ata port 7, drive bay 3, initiator 1
ledctl: AMD SGPIO Header: 00100030
ledctl:            message type: 3                 data size: 0   
ledctl:            message size: 10  
ledctl: AMD SGPIO Request Register: 00c08240 00000001
ledctl:              frame type: 40                 function: 82  
ledctl:           register type: c0           register index: 0   
ledctl:          register count: 1   
ledctl: AMD SGPIO AMD Register: 00000061
ledctl:               initiator: 1                  polarity: 0   
ledctl:           bypass enable: 1          return to normal: 1   
ledctl: CFG SGPIO Header: 00140030
ledctl:            message type: 3                 data size: 0   
ledctl:            message size: 14  
ledctl: CFG SGPIO Request Register: 00008240 00000002
ledctl:              frame type: 40                 function: 82  
ledctl:           register type: 0            register index: 0   
ledctl:          register count: 2   
ledctl: CFG SGPIO Configuration Register: 00800000 00210700
ledctl:                 version: 0         gp register count: 0   
ledctl:      cfg register count: 0              gpio enabled: 1   
ledctl:             drive count: 0          blink gen rate A: 7   
ledctl:        blink gen rate B: 0        force activity off: 2   
ledctl:         max activity on: 1      stretch activity off: 0   
ledctl:     stretch activity on: 0   
ledctl: TX SGPIO Header: 00100030
ledctl:            message type: 3                 data size: 0   
ledctl:            message size: 10  
ledctl: TX SGPIO Request Register: 00038240 00000001
ledctl:              frame type: 40                 function: 82  
ledctl:           register type: 3            register index: 0   
ledctl:          register count: 1   
ledctl: TX SGPIO TX Register: a0a0a0a0
ledctl: 	drive 0: error 0, locate 0, activity 5
ledctl: 	drive 1: error 0, locate 0, activity 5
ledctl: 	drive 2: error 0, locate 0, activity 5
ledctl: 	drive 3: error 0, locate 0, activity 5
ledctl: 
<snip>
```

mtkaczyk, 2024-06-20:
Hello,
This is AMD issue, so I'm adding @nfont.

Intel does not have AMD hardware to support you, sorry!
Thanks,
Mariusz

minorsatellite, 2024-06-20:
FYI, ledmon/ledctl has never worked on my Dell AMD system. I have been told its an edge case has having to do with my storage enclosure but I am highly skeptical of that view. I had to disable it because it was clogging syslog with unending stream of error messages.

nfont, 2024-06-25:
I should have some time this week to take a look at this.

ktanska, 2024-09-06:
@nfont were you able to look at this?

mtkaczyk, 2024-10-04:
No response from the AMD team. There is no one else I can ask for support. I have to close this, sorry :(
@nfont please fell free to reopen if you will find time for this.

closed 2024-10-04 by mtkaczyk