md-raid-utilities/ledmon #246: [BUG]: locate E3.S NVMe SSD show many abnormal output (SCSI: Unable to locate...)

jackeichen, 2024-08-27:
### Description

When try to locate the nvme disk, get many abnormal output.

### Steps to reproduce bug

1. start ledmon.serivce
2. locate the nvme disk: 
    `# ledctl locate=/dev/nvme2n1`

### Expected behavior

get many abnormal output:

```
[root@localhost ~]# ledctl locate=/dev/nvme2n1
ledctl: SCSI: Unable to locate slot in enclosure 0
ledctl: SCSI: Unable to locate slot in enclosure 1
ledctl: SCSI: Unable to locate slot in enclosure 2
ledctl: SCSI: Unable to locate slot in enclosure 3
ledctl: SCSI: Unable to locate slot in enclosure 4
ledctl: SCSI: Unable to locate slot in enclosure 5
ledctl: SCSI: Unable to locate slot in enclosure 6
ledctl: SCSI: Unable to locate slot in enclosure 7
ledctl: SCSI: Unable to locate slot in enclosure 8
ledctl: SCSI: Unable to locate slot in enclosure 9
ledctl: SCSI: Unable to locate slot in enclosure 10
ledctl: SCSI: Unable to locate slot in enclosure 11
ledctl: SCSI: Unable to locate slot in enclosure 12
ledctl: SCSI: Unable to locate slot in enclosure 13
ledctl: SCSI: Unable to locate slot in enclosure 0
ledctl: SCSI: Unable to locate slot in enclosure 1
ledctl: SCSI: Unable to locate slot in enclosure 2
ledctl: SCSI: Unable to locate slot in enclosure 3
ledctl: SCSI: Unable to locate slot in enclosure 4
ledctl: SCSI: Unable to locate slot in enclosure 5
ledctl: SCSI: Unable to locate slot in enclosure 6
ledctl: SCSI: Unable to locate slot in enclosure 7
ledctl: SCSI: Unable to locate slot in enclosure 8
ledctl: SCSI: Unable to locate slot in enclosure 9
ledctl: SCSI: Unable to locate slot in enclosure 10
ledctl: SCSI: Unable to locate slot in enclosure 11
ledctl: SCSI: Unable to locate slot in enclosure 12
ledctl: SCSI: Unable to locate slot in enclosure 13
ledctl: SCSI: Unable to locate slot in enclosure 0
ledctl: SCSI: Unable to locate slot in enclosure 1
ledctl: SCSI: Unable to locate slot in enclosure 2
ledctl: SCSI: Unable to locate slot in enclosure 3
ledctl: SCSI: Unable to locate slot in enclosure 4
ledctl: SCSI: Unable to locate slot in enclosure 5
ledctl: SCSI: Unable to locate slot in enclosure 6
ledctl: SCSI: Unable to locate slot in enclosure 7
ledctl: SCSI: Unable to locate slot in enclosure 8
ledctl: SCSI: Unable to locate slot in enclosure 9
ledctl: SCSI: Unable to locate slot in enclosure 10
ledctl: SCSI: Unable to locate slot in enclosure 11
ledctl: SCSI: Unable to locate slot in enclosure 12
ledctl: SCSI: Unable to locate slot in enclosure 13
ledctl: SCSI: Unable to locate slot in enclosure 0
ledctl: SCSI: Unable to locate slot in enclosure 1
ledctl: SCSI: Unable to locate slot in enclosure 2
ledctl: SCSI: Unable to locate slot in enclosure 3
ledctl: SCSI: Unable to locate slot in enclosure 4
ledctl: SCSI: Unable to locate slot in enclosure 5
ledctl: SCSI: Unable to locate slot in enclosure 6
ledctl: SCSI: Unable to locate slot in enclosure 7
ledctl: SCSI: Unable to locate slot in enclosure 8
ledctl: SCSI: Unable to locate slot in enclosure 9
ledctl: SCSI: Unable to locate slot in enclosure 10
ledctl: SCSI: Unable to locate slot in enclosure 11
[... 141 lines cut ...]
/sys/devices/pci0000:f2/0000:f2:00.5 (VMD)
/sys/devices/pci0000:37/0000:37:02.0/0000:38:00.0/0000:39:02.0 (NPEM)
/sys/devices/pci0000:8a/0000:8a:02.0/0000:8b:00.0/0000:8c:03.0 (NPEM)
/sys/devices/pci0000:b6/0000:b6:02.0/0000:b7:00.0/0000:b8:02.0 (NPEM)
/sys/devices/pci0000:37/0000:37:02.0/0000:38:00.0/0000:39:1f.0/0000:3f:00.0 (SCSI)
/sys/devices/pci0000:0a/0000:0a:02.0/0000:0b:00.0/0000:0c:02.0 (NPEM)
/sys/devices/pci0000:75/0000:75:00.5/pci10001:80/10001:80:06.0 (NPEM)
/sys/devices/pci0000:b6/0000:b6:02.0/0000:b7:00.0/0000:b8:1f.0/0000:be:00.0 (SCSI)
/sys/devices/pci0000:6c/0000:6c:02.0/0000:6d:00.0 (SCSI)
/sys/devices/pci0000:37/0000:37:02.0/0000:38:00.0/0000:39:01.0 (NPEM)
/sys/devices/pci0000:0a/0000:0a:00.5 (VMD)
/sys/devices/pci0000:8a/0000:8a:02.0/0000:8b:00.0/0000:8c:02.0 (NPEM)
/sys/devices/pci0000:b6/0000:b6:02.0/0000:b7:00.0/0000:b8:01.0 (NPEM)
/sys/devices/pci0000:75/0000:75:00.5/pci10001:80/10001:80:02.0 (NPEM)
/sys/devices/pci0000:37/0000:37:02.0/0000:38:00.0/0000:39:04.0 (NPEM)
/sys/devices/pci0000:0a/0000:0a:02.0/0000:0b:00.0/0000:0c:01.0 (NPEM)
/sys/devices/pci0000:b6/0000:b6:02.0/0000:b7:00.0/0000:b8:04.0 (NPEM)
/sys/devices/pci0000:f2/0000:f2:00.5/pci10002:00/10002:00:06.0 (NPEM)
/sys/devices/pci0000:0a/0000:0a:02.0/0000:0b:00.0/0000:0c:04.0 (NPEM)
/sys/devices/pci0000:75/0000:75:00.5 (VMD)
/sys/devices/pci0000:37/0000:37:02.0/0000:38:00.0/0000:39:00.0 (NPEM)
/sys/devices/pci0000:8a/0000:8a:02.0/0000:8b:00.0/0000:8c:01.0 (NPEM)
/sys/devices/pci0000:b6/0000:b6:02.0/0000:b7:00.0/0000:b8:00.0 (NPEM)
/sys/devices/pci0000:37/0000:37:02.0/0000:38:00.0/0000:39:03.0 (NPEM)
```

jackeichen, 2024-08-27:
Addtional information：

1. Although E3.S NVMe SSD show many abnormal output, the led blinks (works as expected);
2. The test could pass in U.2 NVMe SSD, but failed in E3.S NVMe SSD;

bkucman, 2024-08-30:
Hi @jackeichen, 
thanks for reporting the issue, this noisy error message comes from the initialization stage of block devices in ledctl, so it has no effect on setting the LED status on the given nvme disk, as it is a different controller.

Could you check if setting the LED status on SATA drives works? It seems that it is from this controller these error messages come.
`lrwxrwxrwx. 1 root root 0 Aug 27 02:32 sda -> ../devices/pci0000:6c/0000:6c:02.0/0000:6d:00.0/host0/target0:2:2/0:2:2:0/block/sda`
`lrwxrwxrwx. 1 root root 0 Aug 27 02:32 sdb -> ../devices/pci0000:6c/0000:6c:02.0/0000:6d:00.0/host0/target0:2:0/0:2:0:0/block/sdb`

Please also provide logs from the command with increased logging level and SCSI/SATA controller information:
- `ledctl --log-level=all locate=/dev/nvme2n1`
- `ledctl --log-level=all --list-controllers` 
- `lspci -vv`

Thanks,
Blazej

mtkaczyk, 2024-08-30:
@jackeichen please also let us know, in with ledmon version had it been introduced. I see that migration to lib didn't changed it:
https://github.com/intel/ledmon/commit/38ac67fc5b85d4a85a49a9274e387ffd095e7516#diff-9f162c028bce3fdebfebdc7089698c976cc43c8a49e6b04b6feb8fc2137ce4eb

jackeichen, 2024-09-14:
@mtkaczyk   @ktanska    hi guys, my server is running test, until the next week. Could you wait some days？

ktanska, 2024-09-18:
Hi @jackeichen 
OK, please let us know when the tests finished and attach logs to analyze it.

jackeichen, 2024-09-19:
> Hi @jackeichen, thanks for reporting the issue, this noisy error message comes from the initialization stage of block devices in ledctl, so it has no effect on setting the LED status on the given [...]

Hi bkucman @bkucman , here is your logs.
[lspci_vv.txt](https://github.com/user-attachments/files/17054941/lspci_vv.txt)
[ledctl_log_level_all.txt](https://github.com/user-attachments/files/17054942/ledctl_log_level_all.txt)

jackeichen, 2024-09-19:
> @jackeichen please also let us know, in with ledmon version had it been introduced. I see that migration to lib didn't changed it: [38ac67f#diff-9f162c028bce3fdebfebdc7089698c976cc43c8a49e6b04b6feb8fc2137ce4eb](https://github.com/intel/ledmon/commit/38ac67fc5b85d4a85a49a9274e387ffd095e7516#diff-9f162c028bce3fdebfebdc7089698c976cc43c8a49e6b04b6feb8fc2137ce4eb)

Did you mean this?
```
[root@localhost ~]# ledmon -v
Intel(R) Enclosure LED Monitor Service 1.0.0
Copyright (C) 2009-2024 Intel Corporation.

ledmon[21905]: exit status is STATUS_SUCCESS.
```

mtkaczyk, 2024-10-04:
Hi @jackeichen 
For the issue that you reported initially, aabour SCSI verbose logging, this one might help:

jackeichen, 2024-11-07:
Looks great！ Close this issue.

closed 2024-11-07 by jackeichen