md-raid-utilities/ledmon #249: [QUESTION] Can non-VMD NVMe drives be supported?

mattventura, 2024-09-06:
### Question

I was trying out a setup with an X11SRM-VF attached to a BPN-SAS3-826EL1-N4 via the OCuLink connectors on the motherboard. I was having some trouble getting the LEDs to work correct with VMD - the amber "do not remove" LEDs would stay on, except when using the "locate" feature in the BIOS, which would just turn them off rather than blinking the red LED. Since the behavior is buggy in the BIOS as well, I don't think this is an ledctl bug.

I turned VMD off, and I was able to control the backplane LEDs by doing `echo 1 > /sys/bus/pci/slots/X/attention`, since a quick perusal of the ledctl source code and an strace revealed that that is what ledctl does under the hood when using an NVMe drive. I created [a script](https://gist.github.com/mattventura/159a2a648996c2273ddff43e6b185c3f) to perform the same function as ledctl for non-VMD drives. This also worked when using an AOC-SLG3-4E2P PCIe switch card.

My question is - given that LED management seems to be possible for non-VMD drives as long as the PCIe slot has the 'attention' property (and of course, is wired up in the underlying hardware), is there a particular reason this can't be supported in ledctl?

### Environment

Include relevant details about the environment you're working in.
OS: ProxMox VE, kernel 6.8.4-2-pve
What type of controller and disk the question is directed to:
Onboard OCuLink ports on an X11SRM-VF (as well as an AOC-SLG3-4E2P), tested with a Kioxia CM-5 and HGST SN200.

mtkaczyk, 2024-09-06:
> My question is - given that LED management seems to be possible for non-VMD drives as long as the PCIe slot has the 'attention' property (and of course, is wired up in the underlying hardware), is there a particular reason this can't be supported in ledctl?

This is not a part of standard PCIe specification. Simply, to know what you can set you must know what hardware you are referring to. For that reason implementation is limited to VMD and it is divided into vmdssd.c (the device with led support) and pci_slot.c (the device used to led control).
Pull requests are welcomed, but please note that there are better LED solutions. NPEM/DSM has been recently accepted in kernel:
https://lore.kernel.org/linux-pci/Ztlt_ZUy1EH1VQgC@smile.fi.intel.com/T/#t

I believe that more and more hardware devices are going to support NPEM in the future.

mtkaczyk, 2024-09-06:
Mention: #250

mtkaczyk, 2024-09-23:
Hey @mattventura 
Do you have more questions? Can I close this?

mtkaczyk, 2024-10-04:
No response, closing!

closed 2024-10-04 by mtkaczyk