md-raid-utilities/ledmon #255: [QUESTION] Not working on DL380e G8

TMLKyza, 2024-10-20:
### Question

Good evening to everyone,
I'm trying to set up my machine to use ledmon, with the goal of writing a script to highlight a non-healthy disk in my ZFS pool.

Currently machine uses 2 controllers:
```
❯ sudo ledctl --list-controllers
/sys/devices/pci0000:00/0000:00:1f.2 (AHCI)
/sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0 (SCSI)
```
on the first one I've attached 2 drives for the mdadm boot drive and a backup drive for the main ZFS pool.
On the latter instead there is attached the backplane of my DL380e G8 with the 6 disks populating the zpool.

Now my problem is that whenever I try (both using disk/by-id and /dev/sdX) to turn on any led nothing basically happens.
Googling didn't find anything of interest other than my SAS controller should be supported.

I also edited the systemd unit by adding `--log=/var/log/ledmon.log --all --config=/etc/ledmon.conf` this way I was finally able to get some logging going and blacklisted the controller that I cant really use (its just a SAS to SATA splitter).
This is what i get whenever I startup the unit:
```
❯ cat /var/log/ledmon.log
Oct 20 15:36:54    INFO: monitor service has been started...
Oct 20 15:36:54   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:36:54   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:36:54    INFO: NEW /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:3/end_device-0:3/target0:0:3/0:0:3:0/block/sdd: state 'Oneshot Off'.
Oct 20 15:36:54    INFO: NEW /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:1/end_device-0:1/target0:0:1/0:0:1:0/block/sdb: state 'Oneshot Off'.
Oct 20 15:36:54    INFO: NEW /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:5/end_device-0:5/target0:0:5/0:0:5:0/block/sdg: state 'Oneshot Off'.
Oct 20 15:36:54    INFO: NEW /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:4/end_device-0:4/target0:0:4/0:0:4:0/block/sde: state 'Oneshot Off'.
Oct 20 15:36:54    INFO: NEW /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:2/end_device-0:2/target0:0:2/0:0:2:0/block/sdc: state 'Oneshot Off'.
Oct 20 15:36:54    INFO: NEW /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:0/end_device-0:0/target0:0:0/0:0:0:0/block/sda: state 'Oneshot Off'.
```

and this is what logs whenever i try to blink a device:
```
❯ sudo ledctl locate=/dev/sda
ledctl: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
ledctl: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
```
Finally this is what gets appended in the logfile:
```
Oct 20 15:37:04   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:37:04   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:37:04    INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:3/end_device-0:3/target0:0:3/0:0:3:0/block/sdd: from 'Oneshot Off' to 'None'.
Oct 20 15:37:04    INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:1/end_device-0:1/target0:0:1/0:0:1:0/block/sdb: from 'Oneshot Off' to 'None'.
Oct 20 15:37:04    INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:5/end_device-0:5/target0:0:5/0:0:5:0/block/sdg: from 'Oneshot Off' to 'None'.
Oct 20 15:37:04    INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:4/end_device-0:4/target0:0:4/0:0:4:0/block/sde: from 'Oneshot Off' to 'None'.
Oct 20 15:37:04    INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:2/end_device-0:2/target0:0:2/0:0:2:0/block/sdc: from 'Oneshot Off' to 'None'.
Oct 20 15:37:04    INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:0/end_device-0:0/target0:0:0/0:0:0:0/block/sda: from 'Oneshot Off' to 'None'.
Oct 20 15:37:14   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:37:14   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:37:24   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:37:24   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:37:34   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:37:34   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:37:44   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:37:44   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:37:54   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:37:54   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 20 15:38:04   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:38:04   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:38:15   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:38:15   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:38:25   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:38:25   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:38:35   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:38:35   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:38:45   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:38:45   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:38:55   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:38:55   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 20 15:39:00   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:39:00   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 20 15:39:05   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on blacklist, ignoring
Oct 20 15:39:05   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
```

To me it looks like its not able to access the correct backplane, but I'm quite lost to be fair.

Let me know if you need further information and thanks for the time!

### Environment

Working with ZFS and mdadm
OS: Debian 12
What type of controller and disk the question is directed to: The controller is a B320i but should be disabled, in lscpi i see the SAS controller which is Broadcom / LSI SAS2308 PCI-Express Fusion-MPT SAS-2

TMLKyza, 2024-10-24:
Hi, thanks for your reply.
I indeed already blacklisted the AHCI device, see:
```
❯ sudo cat /etc/ledmon.conf
# Exclude SATA Controllers, Only use shelf HBA
EXCLUDELIST=/sys/devices/pci0000:00/0000:00:1f.2
```
I've pulled and compiled version 1 from the release and tried what you sent me but all i get is:
```
❯  sudo ledctl --get-slot --device /dev/sdc --controller-type=SCSI
ledctl: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
ledctl: Slot was not found for provided parameters.
❯  sudo ledctl --set-slot --device /dev/sdc --controller-type=SCSI
ledctl: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
ledctl: Invalid IBPI state in the request.
```

I've reloaded the systemd units and restarted ledmon unit but still nothing.

As before, logs report these lines:
```
Oct 24 14:45:20   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:45:20   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:45:23   DEBUG: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
Oct 24 14:45:23   DEBUG: (_set_block_state): device: sda, state: NORMAL
Oct 24 14:45:23   DEBUG: (_set_block_state): device: sdg, state: NORMAL
Oct 24 14:45:24   ERROR: Invalid IBPI state in the request.
Oct 24 14:45:30   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:45:31   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:45:41   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:45:41   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:45:51   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:45:51   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:46:01   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:46:01   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:46:11   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:46:11   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:46:21   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:46:21   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:46:32   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:46:32   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:46:42   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:46:42   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:46:52   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:46:52   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:47:02   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:47:02   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:47:12   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:47:12   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
Oct 24 14:47:22   DEBUG: /sys/devices/pci0000:00/0000:00:1f.2 found on EXCLUDELIST, ignoring
Oct 24 14:47:22   DEBUG: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
```
Again, sorry for the inconvenience and thanks for your time

ktanska, 2024-10-24:
> I've pulled and compiled version 1 from the release and tried what you sent me but all i get is:
❯  sudo ledctl --get-slot --device /dev/sdc --controller-type=SCSI
ledctl: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
ledctl: Slot was not found for provided parameters.

From this output I can see that /dev/sdc is not visible for ledctl under SCSI controller. Maybe something is wrong with your configuration? You can list available slots for SCSI controller using:
`ledctl --list-slots --controller-type=SCSI`

> ❯  sudo ledctl --set-slot --device /dev/sdc --controller-type=SCSI
> ledctl: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
> ledctl: (_set_block_state): device: sda, [...]

Sorry, I did mistake in this command, it should contain IBPI state too:
`ledctl --set-slot --device /dev/sda --controller-type=SCSI --state=locate`
But I suspect it will not work, and return the same error as "set-slot" command.

TMLKyza, 2024-10-24:
Yeah you are totally right it cant see any slots under SCSI:
```
❯ sudo ledctl --set-slot --device /dev/sdc --controller-type=SCSI --state=locate
ledctl: (raid_device_init) path: md127, level=2, state=7, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
ledctl: Slot was not found for provided parameters.
❯ sudo ledctl --list-slots --controller-type=SCSI
ledctl: (raid_device_init) path: md127, level=2, state=6, degraded=0, disks=2, type=1
ledctl: (_set_block_state): device: sda, state: NORMAL
ledctl: (_set_block_state): device: sdg, state: NORMAL
```
Why is that tho ? if i look at the `--list-controllers` it shows up and in the logs you can ever read lines like:
`INFO: CHANGE /sys/devices/pci0000:00/0000:00:01.1/0000:07:00.0/host0/port-0:2/end_device-0:2/target0:0:2/0:0:2:0/block/sdc: from 'Oneshot Off' to 'None'.`

ktanska, 2024-10-24:
Hi @tasleson can you please help here?
I suspect that LEDs for SCSI are not turned on. Is it configurable on platform?

tasleson, 2024-10-24:
The SCSI led support in ledmon is SES (SCSI Enclosure Services).  Not all systems with a SCSI adapter and disks will support this.  To get additional data debug data, I would do the following:

See if you have an enclosure, 

```bash
# lsscsi -tg
```
You should get something like:
```
# lsscsi -tg
...
[10:1:0:0]   disk    sas:0x0000000000000000          /dev/sda   /dev/sg1 
[12:0:0:0]   disk    sas:0x500015535990a200          /dev/sdb   /dev/sg2 
[12:0:1:0]   disk    sas:0x500015535990a201          /dev/sdc   /dev/sg3 
[12:0:2:0]   disk    sas:0x500015535990a202          /dev/sdd   /dev/sg4 
[12:0:3:0]   disk    sas:0x500015535990a203          /dev/sde   /dev/sg5 
[12:0:4:0]   disk    sas:0x500015535990a204          /dev/sdf   /dev/sg6 
[12:0:5:0]   disk    sas:0x500015535990a205          /dev/sdg   /dev/sg7 
[12:0:6:0]   disk    sas:0x500015535990a206          /dev/sdh   /dev/sg8 
[12:0:7:0]   disk    sas:0x500015535990a207          /dev/sdi   /dev/sg9 
[12:0:8:0]   disk    sas:0x500015535990a208          /dev/sdj   /dev/sg10
[12:0:9:0]   disk    sas:0x500015535990a209          /dev/sdk   /dev/sg11
[12:0:10:0]  disk    sas:0x500015535990a20a          /dev/sdl   /dev/sg12
[12:0:11:0]  disk    sas:0x500015535990a20b          /dev/sdm   /dev/sg13
[12:0:12:0]  disk    sas:0x500015535990a20c          /dev/sdn   /dev/sg14
[12:0:13:0]  disk    sas:0x500015535990a20d          /dev/sdo   /dev/sg15
[12:0:14:0]  disk    sas:0x50014ee5aaab876e          /dev/sdp   /dev/sg16
[12:0:15:0]  disk    sas:0x500015535990a20f          /dev/sdq   /dev/sg17
[12:0:16:0]  enclosu sas:0x500015535990a23e          -          /dev/sg18
...
```
The important part is you find 1 or more lines which are  **enclosu**

Assuming you find at least 1 enclosure, then I would try the following, you use the SCSCI generic device, eg. `/dev/sgN`

```bash
# sg_ses /dev/sg18
  PROMISE   3U-SAS-16-D BP    0107
Supported diagnostic pages diagnostic page:
  Supported Diagnostic Pages [sdp] [0x0]
  Configuration (SES) [cf] [0x1]
  Enclosure Status/Control (SES) [ec,es] [0x2]
  Threshold In/Out (SES) [th] [0x5]
  Enclosure Busy (SES-2) [eb] [0x9]
  Additional Element Status (SES-2) [aes] [0xa]
```

Depending on the hardware support you have, you'll see different information here.   What I usually do next is the following.

Figure out what slot a drive is in, you need to use sas address, eg. /dev/sdb has sas address 0x500015535990a200 (refer to lsscsi -tg output)

```
# sg_ses --join /dev/sg18 | grep -B 17 0x500015535990a200
[0,0]  Element type: Array device slot
  Enclosure Status:
    Predicted failure=0, Disabled=0, Swap=0, status: OK
    OK=0, Reserved device=0, Hot spare=0, Cons check=0
    In crit array=0, In failed array=0, Rebuild/remap=0, R/R abort=0
    App client bypass A=0, Do not remove=0, Enc bypass A=0, Enc bypass B=0
    Ready to insert=0, RMV=0, Ident=1, Report=0
    App client bypass B=0, Fault sensed=0, Fault reqstd=0, Device off=0
    Bypassed A=0, Bypassed B=0, Dev bypassed A=0, Dev bypassed B=0
  Additional Element Status:
    Transport protocol: SAS
    number of phys: 1, not all phys: 1, device slot number: 1
    phy index: 0
      SAS device type: no SAS device attached
      initiator port for:
      target port for: SATA_device
      attached SAS address: 0x500015535990a23f
      SAS address: 0x500015535990a200
```
We can see from this that the drive is in **"device slot number: 1"**, lets retrieve it's status.

```
# sg_ses --dev-slot-num=1 --get=ident /dev/sg18
1
```
Which indicates the IDENT LED is on, lets compare to ledctl

```
# ledctl --list-slots --controller-type=SCSI | grep /dev/sdb
/dev/shm/ledmon.conf: does not exist, using global config file
/etc/ledmon.conf: does not exist, using built-in defaults
slot: sg18-0          led state: LOCATE          device: /dev/sdb
```
Lets set to normal

```
# sg_ses --dev-slot-num=1 --set=ident=0 /dev/sg18
```

Check with ledctl
```
# ledctl --list-slots --controller-type=SCSI | grep /dev/sdb
/dev/shm/ledmon.conf: does not exist, using global config file
/etc/ledmon.conf: does not exist, using built-in defaults
slot: sg18-0          led state: NORMAL          device: /dev/sdb
```

For much more information about using SES and sg_ses, see: https://sg.danny.cz/sg/sg_ses.html

If you find that you do have support for SES and that sg_ses works with it, please write up an issue and include all the data you dump from sg_ses, thanks.

TMLKyza, 2024-10-24:
Many thanks for the effort, I think we found the culprit:
```
❯ sudo lsscsi -tg
[0:0:0:0]    disk    sas:0x4433221106000000          /dev/sdb   /dev/sg0 
[0:0:1:0]    disk    sas:0x5000cca06d109a45          /dev/sdc   /dev/sg1 
[0:0:2:0]    disk    sas:0x5000cca06d139ee5          /dev/sdd   /dev/sg2 
[0:0:3:0]    disk    sas:0x4433221104000000          /dev/sde   /dev/sg3 
[0:0:4:0]    disk    sas:0x4433221105000000          /dev/sdf   /dev/sg4 
[0:0:5:0]    disk    sas:0x4433221107000000          /dev/sdh   /dev/sg5 
[1:0:0:0]    disk    sata:5000c50020ac5110           /dev/sda   /dev/sg6 
[2:0:0:0]    disk    sata:5000cca583c35d49           /dev/sdg   /dev/sg7 
[3:0:0:0]    disk    sata:50014ee260083734           /dev/sdi   /dev/sg8 
```
This means that my controller is not compatible with ledmon from what I understand, so what are my options next ? 
would using a P420 in HBA mode be the solution? 
Are there any other alternatives to control leds ? (i would prefer to keep the pcie slot free).
I don't really need all the complicated stuff that HP does I just need a way to identify dead drives from drive ids.

Regardless, huge thanks for your time again.

tasleson, 2024-10-24:
> This means that my controller is not compatible with ledmon from what I understand, so what are my options next ?

It's not that the controller doesn't support SES, it's that your hardware doesn't appear to support SES.  The LEDs on the actual disk drive are not the ones we are utilizing with SES.  These are the LEDs that are on the drive sleds or on the back plane.  You can toggle the LEDs on the drive sleds/back plane without even having a disk drive in the slot.  I don't know what your HP hardware has, or what it supports.

> would using a P420 in HBA mode be the solution?

It looks like whatever you're using is in HBA mode, otherwise I don't think you would see the individual drives.

> Are there any other alternatives to control leds ? (i would prefer to keep the pcie slot free).

Check your hardware docs, and look around in the iLO.  Some HP hardware allows you to turn on/off identification LEDs in the iLO.

> I don't really need all the complicated stuff that HP does I just need a way to identify dead drives from drive ids.

Identifying storage devices in a system and locating them is often wanted, but not always available in a standards compliant way.  You may need to use the `ssacli` tool to work with LEDs.

TMLKyza, 2024-10-24:
> Identifying storage devices in a system and locating them is often wanted, but not always available in a standards compliant way. You may need to use the ssacli tool to work with LEDs.

`ssacli` is what i was trying to stay away from, as its quite useless in this case, to blink a led i need to know the position in the enclosure.

>It looks like whatever you're using is in HBA mode, otherwise I don't think you would see the individual drives.
It's actually a bit more complicated than this. Usually these servers ship with the P420, that I removed because I need as many PCI slots as possible for GPUs and because I use ZFS.
But this server also has a B320i which its a sort of RAID controller but its software RAID (it's a strange thing to be fair) and needs a license (in full HP style).
However you can disable this card and use the stock backplane which is:
```
❯ lspci
...
07:00.0 Serial Attached SCSI controller: Broadcom / LSI SAS2308 PCI-Express Fusion-MPT SAS-2 (rev 01)
...
```
And from what I read online this should support SGPIO, that's why I was hoping it could work.

tasleson, 2024-10-24:
> And from what I read online this should support SGPIO, that's why I was hoping it could work.

If you do have SGPIO, look into `sgpio` utility,

TMLKyza, 2024-10-24:
Will do that, thanks a lot for the help! Have a great week!