md-raid-utilities/ledmon #266: Call for Community Support & Possible Fork of ledmon

tasleson, 2025-09-15:
Tagging past contributors and maintainers who may have insight or interest:

@mtkaczyk @apaszkie @ldorau @mdabrows @mzylowski @ktanska @mpatelcz @bkucman @mgrzonka @pawpiatko @rsobans @nfont @kerolasa @pbaldysi @mku514k @jharg @krzysztofsmolinski @phawryle @dancesWithMachines @njcrawford @oshchirs @woodrow-shen @bigunclemax @lflorcza @djbw @asztukow @komeP @p-alik @alpinus4 @cerby-csy @r08ertliu @ColinIanKing @AdrianBunk @ffontaine @peluse @tpetazzoni 

It appears that Intel, as a company, is no longer actively maintaining the `ledmon` project. However, many Intel engineers, and numerous other community contributors, have been essential to making `ledmon` a reliable and widely used tool, and their efforts are greatly appreciated. Since many of us still rely on `ledmon` for monitoring and managing storage device LEDs, I'd like to open a discussion on how we can keep the project moving forward.

**Key points for discussion:**

- **Testing hardware:** Access to the right hardware is critical. If you have systems where `ledmon` is used, please consider helping with validation and regression testing. Consistent hardware testing before merging changes is one of the most important factors in ensuring the long-term stability, quality, and usefulness of the project.
- **Ongoing contributions:** Past contributors, your experience is invaluable here. Even small fixes, reviews, or documentation updates can help.
- **Project home:** If upstream maintenance is no longer viable on the `intel/ledmon` github repository, we need to decide where a community-driven fork should live. Options include GitHub, GitLab, or another open platform. Whatever location we choose, I believe it should be hardware vendor neutral so that no single vendor dictates the project's direction. A neutral home will help encourage broader participation, attract contributions from multiple organizations, and ensure the tool remains useful across diverse hardware environments.
- **Diversified administration:** To avoid single points of failure, any fork should have a set of maintainers from different organizations and backgrounds. This helps ensure continuity, shared responsibility, and broader testing coverage.

👉 If you've contributed before or depend on `ledmon` in your environment, please share your thoughts below. Together we can decide on next steps to keep the project alive and supported.

mtkaczyk, 2025-09-16:
Hi @tasleson 
These folks:
@mtkaczyk  @apaszkie @ldorau @mdabrows @mzylowski @ktanska @mpatelcz @bkucman @mgrzonka @pawpiatko @rsobans @pbaldysi @mku514k @krzysztofsmolinski @phawryle @dancesWithMachines @oshchirs  @lflorcza @djbw @asztukow @alpinus4 @peluse 
worked on ledmon during their adventure at Intel. Franky speaking, I don't think they will be interested working in spare time on  ledmon today. So far I know only 5-6 ppl form the list are still at Intel these days.

I think you can count on @nfont (AMD) and maybe @jharg (Supermicro?). 
Other folks are one or two times contributors. I will try to be around and help with review.

peluse, 2025-09-18:
@tasleson @mtkaczyk speaking for Intel, this is a time of transition.  This will likely not be archived but also not actively monitored or handed off for a little while.  @mtkaczyk if you don't mind keeping an eye out for a bit we'll get this properly handed off within the next few months...

k0ste, 2026-02-06:
> As for the new release, I expect some changes soon, if they don't appear, I will tag a new release.

Almost quarter has passed. Last tag from 2024, now is year 2026 👀